gnuradio/gnuradio #2297: Add C++ support to all blocks

haakov, 2019-01-07:
The checklists below indicate which blocks have C++ support in master. (**lots** of text below)

<details>
<summary>gr-analog</summary>

- [ ] analog_agc2_xx
- [ ] analog_agc3_xx
- [ ] analog_agc_xx
- [ ] analog_am_demod_cf
- [ ] analog_const_source_x
- [ ] analog_cpfsk_bc
- [ ] analog_ctcss_squelch_ff
- [ ] analog_dpll_bb
- [ ] analog_fastnoise_source_x
- [ ] analog_feedforward_agc_cc
- [ ] analog_fm_deemph
- [ ] analog_fm_demod_cf
- [ ] analog_fmdet_cf
- [ ] analog_fm_preemph
- [ ] analog_frequency_modulator_fc
- [ ] analog_nbfm_rx
- [ ] analog_nbfm_tx
- [ ] analog_noise_source_x
- [ ] analog_phase_modulator_fc
- [ ] analog_pll_carriertracking_cc
- [ ] analog_pll_freqdet_cf
- [ ] analog_pll_refout_cc
- [ ] analog_probe_avg_mag_sqrd_x
- [ ] analog_pwr_squelch_xx
- [ ] analog_quadrature_demod_cf
- [ ] analog_rail_ff
- [ ] analog_random_source_x
- [ ] analog_random_uniform_source_x
- [ ] analog_sig_source_x
- [ ] analog_simple_squelch_cc
- [ ] analog_standard_squelch
- [ ] analog_wfm_rcv
- [ ] analog_wfm_rcv_pll
- [ ] analog_wfm_tx

</details>

<details>
<summary>gr-audio</summary>

- [ ] audio_sink
- [ ] audio_source

</details>

<details>
<summary>gr-blocks</summary>

- [ ] blocks_abs_xx
- [ ] blocks_add_const_vxx
- [ ] blocks_add_xx
- [ ] blocks_and_const_xx
- [ ] blocks_and_xx
- [ ] blocks_argmax_xx
- [ ] blocks_burst_tagger
- [ ] blocks_char_to_float
- [ ] blocks_char_to_short
- [ ] blocks_complex_to_arg
- [ ] blocks_complex_to_float
- [ ] blocks_complex_to_imag
- [ ] blocks_complex_to_interleaved_char
- [ ] blocks_complex_to_interleaved_short
- [ ] blocks_complex_to_mag
- [ ] blocks_complex_to_magphase
- [ ] blocks_complex_to_mag_squared
[... 466 lines cut ...]
- [ ] zeromq_push_sink
- [ ] zeromq_rep_msg_sink
- [ ] zeromq_rep_sink
- [ ] zeromq_req_msg_source
- [ ] zeromq_req_source
- [ ] zeromq_sub_msg_source
- [ ] zeromq_sub_source

</details>

<details>
<summary>grc</summary>

- [ ] import
- [ ] note
- [ ] options
- [ ] pad_sink
- [ ] pad_source
- [ ] parameter
- [ ] variable
- [ ] variable_config
- [ ] variable_function_probe
- [ ] variable_struct

</details>

michaelld, 2019-01-08:
Can you please explain what you mean by "C++ support in master"? Maybe I'm the only person who doesn't understand, but I doubt it. Having a little background as to what this means and what you're actually doing programming-wise would be useful IMHO, in general & not just for me. Thx!

mbr0wn, 2019-01-10:
Holy :hankey:

michaelld, 2019-01-10:
OK wow this is quite a serious change! Thx for the explanation @mbr0wn & updating the title @noc0lour  ... it all makes more sense now to me!

haakov, 2019-01-10:
Yeah, it was a little vague, sorry about that. This issue is really just to keep track of which blocks are available for C++ generation as @noc0lour said. 

Programming-wise, there are two things that need to be done to add C++ support to a block. 

First, the `cpp` flag will have to be added:

https://github.com/gnuradio/gnuradio/blob/d67fc91544e43ccb3b810ea6b9773b4a18a9be34/gr-blocks/grc/blocks_add_xx.block.yml#L3

(GRC will refuse to generate a C++ flowgraph if it contains blocks that don't have this flag)

Second, the block will need `cpp_templates`:

https://github.com/gnuradio/gnuradio/blob/d67fc91544e43ccb3b810ea6b9773b4a18a9be34/gr-blocks/grc/blocks_add_xx.block.yml#L43-L46

aru31, 2019-01-20:
@haakov can i start working on this issue?

aru31, 2019-01-20:
Can I work on this issue?

ncorgan, 2019-06-01:
gr-dtv WIP here: https://github.com/ncorgan/gnuradio/tree/grc/dtv_cpp

dkozel, 2020-04-05:
@ncorgan Hey Nick, Any chance you can shepherd your gr-dtv additions upstream? Incomplete or not they're a seriously useful contribution and I'd hate to see them bit-rot for another 15 months.

dkozel, 2020-04-05:
https://github.com/gnuradio/gnuradio/pull/2573 covers gr-vocoder. I'll rebase and open a new PR unless you're free to do it @haakov?

ncorgan, 2020-04-06:
Sure. After work, I’ll see what state it’s in and make a rebased pull request for what’s there.

haakov, 2020-04-11:
@dkozel Oh I must've completely missed that somehow. I'll open a new PR.

ncorgan, 2020-04-18:
gr-dtv complete: https://github.com/gnuradio/gnuradio/pull/3388

Jacob2712, 2022-04-23:
So where do I put this info to unlock it??

marcusmueller, 2023-12-22:
Current list of block yamls without cpp template attached. Needs to be manually checked against the list in the first post here, needs a bit of logic to exclude python blocks, and then we can have an updated list. But it's pretty certain that we'll need more developerforce to amend the remaining blocks!
[without_cpp_templates.txt](https://github.com/gnuradio/gnuradio/files/13753895/without_cpp_templates.txt)