gnuradio/gnuradio #2300: Manual build instructions fail

ps-crawford, 2019-01-10:
Trying out the manual build instructions given on the main page:
https://github.com/gnuradio/gnuradio
Using Lubuntu 18.04 64-bit on Intel CPU the following happens:
```
paul@paul-entroware:~/Downloads/software$ git clone https://github.com/gnuradio/gnuradio.git
Cloning into 'gnuradio'...
remote: Enumerating objects: 121, done.
remote: Counting objects: 100% (121/121), done.
remote: Compressing objects: 100% (96/96), done.
remote: Total 141135 (delta 45), reused 43 (delta 25), pack-reused 141014
Receiving objects: 100% (141135/141135), 49.81 MiB | 6.46 MiB/s, done.
Resolving deltas: 100% (109152/109152), done.
paul@paul-entroware:~/Downloads/software$ mkdir build
paul@paul-entroware:~/Downloads/software$ cd build
paul@paul-entroware:~/Downloads/software/build$ cmake -DCMAKE_BUILD_TYPE=Debug ../
CMake Error: The source directory "/home/paul/Downloads/software" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
```
I expected that following the given instructions would cause it to build, not to complain about a missing file.

michaelld, 2019-01-10:
The instructions are:
(1) `git clone https://github.com/gnuradio/gnuradio.git`
(2) `cd gnuradio`
(3) `mkdir build` ...
You missed (2) ...

closed 2019-01-10 by michaelld

ps-crawford, 2019-01-10:
Doh, thanks!