gnuradio/gnuradio #2303: Failing with no error

zhudsonSM, 2019-01-11:
I am trying to use an osmocom source block with the airspy HF+. Everything appears to be installed correctly and doing a "SoapySDRUtil --probe" returns the following:

######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device 

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=AirspyHF
  hardware=AirspyHF
  device_id=0

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 0 Tx
  Timestamps: NO

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CF32
  Native format: CF32 [full-scale=1]
  Antennas: RX
  Full gain range: [0, 0] dB
  Full freq range: [0.009, 31], [60, 260] MHz
    RF freq range: [0.009, 31], [60, 260] MHz
  Sample rates: 0.768 MSps

When attempting to run any flow graph with the osmocom source block I get no error message but the flow graph does not run.

Generating: '/home/zach/top_block.py'

Executing: /usr/bin/python2 -u /home/zach/top_block.py

gr-osmosdr v0.1.4-127-g4d83c606 (0.1.5git) gnuradio 3.7.13.4
built-in source types: file osmosdr fcd rtl rtl_tcp uhd hackrf bladerf rfspace airspy soapy redpitaya 
[INFO] Using format CF32.

>>> Done

Running the exact same flowgraph (I am using simply the source block connected to a QT GUI frequency sink for testing) with an airspy mini works no problem. With no device connected it returns "RuntimeError: SoapySDR::Device::make() no driver specified and no enumeration results" as I would expect.

nickoe, 2019-02-20:
Please format your description properly. See especially the section about multiline code blocks https://help.github.com/articles/basic-writing-and-formatting-syntax/#quoting-code

I can't tell from what you have written that does not work as intended.

zhudsonSM, 2019-02-20:
I was unable to get the osmocom block to work but switched to using the SoapySDR source block and it is now working correctly

closed 2019-02-20 by zhudsonSM