gnuradio/gnuradio #2307: GRC: QtGui: Widgets missing

jdemel, 2019-01-12:
If you run a flowgraph from GRC, e.g. `qtgui_tags_viewing.grc`, The `Qt GUI Range` widget will not show up. Thus, you can't control its value during runtime. It is missing from the generated Python file as well.

This happens with  _3.8tech-preview-231-g2826fd09 (Python 3.6.7)_

haakov, 2019-01-12:
I'll put up a PR for it tomorrow :)

drmpeg, 2019-01-13:
@haakov probably already knows this, but it's a regression in a recent commit. It was working fine not long ago.

[image]

gr-digital/python/digital/utils/tagged_streams.py needs to be fixed also.

-        tag.value = pmt.from_long(length / vlen)
+        tag.value = pmt.from_long(length // vlen)

There's another divide in that file that may need changing.

mbr0wn, 2019-01-14:
Looks like this is fixed.

closed 2019-01-14 by mbr0wn