gnuradio/gnuradio #2327: GRC: hier block with multiple message pad source ports crashes GRC

jdemel, 2019-01-19:
If you generate `/gr-blocks/examples/msg_passing/hier/test_msg_hier.grc` it does produce an invalid `.grc` file which prevents GRC from loading. GRC will complain about non-unique port IDs. If you look at the generated `.grc` file, this is indeed the case.

What makes that issue really annoying is the fact, that once this flowgraph is generated, GRC will crash because it can't load that particular flowgraph.

aru31, 2019-02-15:
Yeah, the thing with GRC, as it crashes because it cannot load that particular flowgraph is really annoying.
@jdemel I think GRC should startup anyway, even if that flowgraph crashes by opening up a recently opened .grc file, whose info we can get from the grc.conf file in .gnuradio folder and also providing some kind of error message on the terminal.

dl1ksv, 2019-03-14:
The easiest way to fix is to change https://github.com/gnuradio/gnuradio/blob/d5e0b4b82aa2f08030d391cc878698295e525b4a/grc/core/platform.py#L180
from 
raise
to
continue

I think it make no sense, to terminate grc completly if only one flowgraph fails to load without errors.

To be done:

Route the error messages to the grc error pane. At the moment you only see error messages in the terminal window you started grc.

But does it make sense to connect two message ( input ) pads to the same input port of a message strobe ?

Nevertheless grc should not terminate.

aru31, 2019-03-18:
I have been working on this issue and the problem I'm facing is that we know that `Messages.send()` will enable us to view error on the GRC terminal but this function is not wroking here and I cannot figure out why.
https://github.com/aru31/gnuradio/blob/f75d15a54e8414703880b1a6566a8c087f7eb337/grc/core/platform.py#L180-L181

PR #2412 (2019-06-19): GRC: Fix for GRC crash in case of flowgraph error
Fixes #2327
closed 2019-06-19 by mbr0wn