gnuradio/gnuradio #2356: Inconsistent documentation for LMS DD equalizer

dernasherbrezon, 2019-02-09:
It looks like documentation is inconsistent to the implementation:
https://github.com/gnuradio/gnuradio/blob/adaa7a265eef17d3b2f9a991e944fe20677b069b/gr-digital/include/gnuradio/digital/lms_dd_equalizer_cc.h#L46

While implementation:
https://github.com/gnuradio/gnuradio/blob/adaa7a265eef17d3b2f9a991e944fe20677b069b/gr-digital/lib/lms_dd_equalizer_cc_impl.cc#L103

Am I missing something?

marcusmueller, 2019-04-21:
I don't think you're missing something. How's the signal performance when you fix the code to do what the docs say, i.e. swap the conj(in) with conj(d_error)?

dernasherbrezon, 2019-04-21:
Sorry, haven't tried. I guess it involves some kind of comparison between ideal mathematical graph and the one produced by equalizer? I don't have enough skills to even approach the problem.

duggabe, 2023-07-14:
Since this block is now deprecated, we should close this issue.

closed 2024-01-03 by marcusmueller