gnuradio/gnuradio #2363: gr-uhd sink segfaults on ARM

drmpeg, 2019-02-18:
I get an occasional segfault from the gr-uhd sink when running on a BeagleBoard X-15 (32-bit Cortex-A15 CPU). It only happens when the flow graph is stopped.

I've traced the problem with gdb, and it appears to be in the async_event_loop() thread. When the flow graph is stopped, UHD will sometimes send the async message BURST_ACK_KEY. async_event_loop() processes this message and tries to create a PMT to send to the message port.

The segfault always happens in the call to dict_add() for the time_spec.

value = pmt::dict_add(value, TIME_SPEC_KEY, time_spec);

Here's the backtrace:
```
(gdb) bt
#0  0x00000006 in ?? ()
#1  0xb677277e in pmt::eqv_raw (x=x@entry=0x23bc818, y=0x23bc840)
    at /home/ubuntu/xfer/gnuradio/gnuradio-runtime/lib/pmt/pmt.cc:1096
#2  0xb6773212 in pmt::assv_raw (obj=0x23bc818, alist=0xaaa0b478)
    at /home/ubuntu/xfer/gnuradio/gnuradio-runtime/lib/pmt/pmt.cc:1227
#3  0xb6773280 in pmt::assv (obj=..., alist=...)
    at /home/ubuntu/xfer/gnuradio/gnuradio-runtime/lib/pmt/pmt.cc:1240
#4  0xb67732dc in pmt::dict_has_key (dict=..., key=...)
    at /home/ubuntu/xfer/gnuradio/gnuradio-runtime/lib/pmt/pmt.cc:929
#5  0xb677495e in pmt::dict_add (dict=..., key=..., value=...)
    at /home/ubuntu/xfer/gnuradio/gnuradio-runtime/lib/pmt/pmt.cc:885
#6  0xac47c658 in gr::uhd::usrp_sink_impl::async_event_loop (this=0x256b9f0)
    at /home/ubuntu/xfer/gnuradio/gr-uhd/lib/usrp_sink_impl.cc:753
#7  0xb66c73d6 in ?? ()
   from /usr/lib/arm-linux-gnueabihf/libboost_thread.so.1.65.1
#8  0xb6e12614 in start_thread (arg=0x5e861a) at pthread_create.c:463
#9  0xb6eca7fc in ?? () at ../sysdeps/unix/sysv/linux/arm/clone.S:73
   from /lib/arm-linux-gnueabihf/libc.so.6
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
(gdb) 
```
The segfault happens in the pmt.cc function eqv_raw when is_integer() is indirectly called. The indirect pointer is bad (the value is 0x6) and the CPU segfaults.

My guess is that this is some race condition exacerbated by the slowness of the ARM processor. The Boost intrusive_ptr stuff is way over my head and I have no idea how to solve this issue. @mbr0wn, any ideas?

noc0lour, 2019-02-18:
oh, that sounds bad. We also have consistent heap-use-after-free related to the intrusive pointer construct if you have a vector of tags and pass it through a flowgraph (vector_source -> block -> vector_sink). And then try to use it from python land after the flowgraph stopped processing.

marcusmueller, 2019-04-21:
since we addressed the intrusive pointers (#2422), is this still an issue?

drmpeg, 2019-04-21:
I will retest.

bhilburn, 2019-06-20:
@drmpeg - Did you ever get a chance to retest this against current `master`?

drmpeg, 2019-06-21:
I couldn't reproduce it. I'm not exactly sure it's fixed with #2422, but it seems to be.

closed 2019-06-21 by drmpeg