gnuradio/gnuradio #2364: gr::sync_decimator seems to eat memory when input has many tags

alexxcons, 2019-02-18:
My input stream has multiple tags. When I use `tag_propagation_policy_t::TPP_ONE_TO_ONE` in my sync_decimator, and two or more tags are in the same "decim" window, than I can observe that my memory climbs up slowly.

If I `pick tag_propagation_policy_t::TPP_ONE_TO_ONE`, than my memory stays constant. (If I do tag propagation "by hand", the memory will increase as well )

My suspicion is, that the problem only arises when multiple tags with the same offset are created on the output ( If I take care to shift tags accordingly, so that each output sample only has one tag, than the memory stays constant. )

I will try to build a simple reproducer when I have time.

gnuradio version: 3.7.12.0

marcusmueller, 2019-04-21:
Hi @alexxcons,

 this sounds serious. Any news on the reproduction? By the way, which system are you on? Maybe we can supply you with a slightly less dated GNU Radio version!

alexxcons, 2019-04-24:
Sorry, currently I dont have time to work on the project (and the reproducer) .. hopefully I will come back to it soon.

I develop via gnuradio @work, on a development-cluster on which I dont have root/sudo. Gnuradio was installed from the distro-package.
So if I want to use a more recent gnuradio version, I will need to compile it from source and install it in my home folder.

marcusmueller, 2019-04-26:
There might elegant ways of doing that, @alexxcons! Is there a "standard" OS running on that cluster?

alexxcons, 2019-05-03:
It's a centos:

> rpm -q centos-release
> centos-release-7-6.1810.2.el7.centos.x86_64

marcusmueller, 2024-01-03:
I think this bug might still be present in 3.10, but it takes investigation to reproduce. (and a 3.7 reproduction won't help).