gnuradio/gnuradio #2365: python3: TypeError: in method file_meta_sink_make

nickoe, 2019-02-18:
I get this error when running a minimal flowgraph with a File Meta Sink block using python3.

```
Traceback (most recent call last):
  File "/tmp/top_block.py", line 123, in <module>
    main()
  File "/tmp/top_block.py", line 111, in main
    tb = top_block_cls()
  File "/tmp/top_block.py", line 77, in __init__
    self.blocks_file_meta_sink_0 = blocks.file_meta_sink(gr.sizeof_gr_complex*1, '/tmp/test.dat', samp_rate, 1, blocks.GR_FILE_FLOAT, True, 1000000, "", False)
  File "/usr/lib/python3.7/site-packages/gnuradio/blocks/blocks_swig0.py", line 1273, in make
    return _blocks_swig0.file_meta_sink_make(*args, **kwargs)
TypeError: in method 'file_meta_sink_make', argument 8 of type 'pmt::pmt_t'
```

[image]

Using these build options:
```
    cmake \
        -DENABLE_GRC=ON \
        -DENABLE_GR_QTGUI=ON \
        -DENABLE_INTERNAL_VOLK=OFF \
        -DCMAKE_INSTALL_PREFIX=/usr \
        -DGR_PYTHON_DIR=/usr/lib/python3.7/site-packages \
        -DQWT_LIBRARIES=/usr/lib/libqwt.so -Wno-dev ../
```

Version 3.8tech-preview-280-g3a533f78, currently latest on master.

nickoe, 2019-02-18:
I sort of feel that these two commits are related to this:
https://github.com/gnuradio/gnuradio/commit/89736b31397330861dcaa550b7891ce6e7a5536f
https://github.com/gnuradio/gnuradio/commit/3c3d23446499313f1351f739a9795cb52454a85e

drmpeg, 2019-02-19:
Verified here. In commit https://github.com/gnuradio/gnuradio/commit/3c3d23446499313f1351f739a9795cb52454a85e, the issue is just covered up by adding a dummy PMT (and import pmt) to the test flow graph.

noc0lour, 2019-02-19:
Is this only an issue with python3? Did not quite follow the discussion on IRC.

nickoe, 2019-02-19:
I am not quite sure yet. I was blocked by my own stupidity. It worked fine with gnuradio 3.7 and python2.

I intend to test that unless someone beats me to it.

noc0lour, 2019-02-19:
That's fine, I suspect it might have something to do with changing the block description to YAML.

nickoe, 2019-02-19:
Ok, I just tested a build of master against python2 and it gives me the same error. So this is not strictly python3 only.

citizenrich, 2019-09-30:
I can reproduce the error when trying to read a detached header file in both Python2 in GR 3.7.13.4 and in Python3 using latest built master.

crespum, 2019-11-07:
Still experiencing this error with GR 3.8.0.0 (Python 3.7.5) when adding a File Meta Sink block. The solution is to use `pmt.make_dict()` in the "Extra Dict." field and add a block importing `pmt`.

petrushy, 2021-01-12:
Hi, this bug seems to be present in gnuradio 3.8.2.0 for windows, installed via conda-forge, python v 3.8.6.

How can one (in gnuradio companion) make a block to import pmt into global namespace before the blocks are initialized and the error occur? "Python snippet" seems to be as a function executed after.

nickoe, 2021-01-13:
It looks like @crespum 's suggested fix has been merged on master at some point along the way, causing 
 https://github.com/gnuradio/gnuradio/issues/3962  meaning that one needs to add the additional import pmt block.

That was really added in https://github.com/gnuradio/gnuradio/commit/89736b31397330861dcaa550b7891ce6e7a5536f, which is also in maint-3.8, and interestingly enough I mentioned that issue as a comment my original report, this could indicate that there could be some issue with the `PYTHONPATH` in the environment when testing.

I don't really know how coanda works, but please check if it makes a difference to set `PYTHONPATH` to the site-packages path of where the gnuradio install is located. 

> How can one (in gnuradio companion) make a block to import pmt into global namespace before the blocks are initialized and the error occur? "Python snippet" seems to be as a function executed after.

I am not sure what you mean with this? I hope you are talking about the exact error message in my original description and not about importing pmt -- if the latter, then this is the wrong issue you are posting on.

willcode, 2021-03-03:
The "extra dict" field is required to be filled at a minimum with an empty dict, and is filled that way by default. An "import pmt" block is required in order to use pmt to specify that field. Updated https://wiki.gnuradio.org/index.php/File_Meta_Sink to include this info.

Block yml could be modified to do the "import pmt" in the yml, and could handle an empty "extra dict" field.

The main question appears to have been answered a few months ago, so closing.

closed 2021-03-03 by willcode