gnuradio/gnuradio #2366: CRC16

michael-roe, 2019-02-18:
(Feature request)

gnuradio has a standard block for CRC32, but not for CRC16. CRC16 with the CCITT polynomial (x**16+x**12+x**5+1) is used in lots of radio protocols (e.g. NATO STANAG 4202, Yaesu System Fusion) so I think it should be included in GNU radios standard library of blocks.

(Ok, I could write a custom block to do this),

noc0lour, 2019-02-19:
Maybe the different CRC implementations could be added as generic FEC decoder/encoder

noc0lour, 2019-02-19:
Moving away from the CRC tagged stream block we currently have

marcusmueller, 2019-04-21:
> (Ok, I could write a custom block to do this),

Code contributions more than welcome!

I generally agree with @noc0lour's sentiment that this would be nice to have in a more general way, but adding a generally usable CRC16 would be a good thing, sooner than later.

daniestevez, 2021-05-29:
Hi! I just stumbled upon this issue. I also have a generic CRC block on the roadmap for gr-satellites, since I keep adding ad-hoc CRC implementations. I can't see to find the time to get to it, but my idea is to have something where the CRC parameters can be specified in the manner of [pycrc](https://pypi.org/project/pycrc/).

For gr-satellites the main use case is checking a CRC at the end of a PDU, but adding a CRC at the end of a PDU is also an important use case.

daniestevez, 2021-05-29:
After [looking at the current CRC32 block](https://github.com/gnuradio/gnuradio/blob/4d306e1f98eb60cf62ca8f0aa59e3939e7826e84/gr-digital/lib/crc32_async_bb_impl.h#L23), I think that an easy way to get a generic CRC block would be to design an interface around `boost::crc_optimal`. Though this is not very good if we want to get away from using boost as a dependency.