gnuradio/gnuradio #2367: python2: invalid syntax for test_modtool.py

nickoe, 2019-02-18:
As I understand it gnuradio 3.8 master branch should support python2 for some time, at last according to https://www.youtube.com/watch?v=Ip_TxZph_-0, but it seems to fail for me.

```
Traceback (most recent call last):
  File "/home/nickoe/gnuradio/build_py2k_bb/python_compile_helper.py", line 6, in <module>
    py_compile.compile(file=src, cfile=gen, doraise=True)
  File "/usr/lib/python2.7/py_compile.py", line 117, in compile
    raise py_exc
py_compile.PyCompileError:   File "/home/nickoe/gnuradio/gr-utils/python/modtool/tests/test_modtool.py", line 110
    print(pylint_stdout.getvalue(), end='')
                                       ^
SyntaxError: invalid syntax

```
I tried to use the same build options as on buildbot.https://ci.gnuradio.org/buildbot/#/builders/50/builds/640/steps/5/logs/stdio
`cmake -DPYTHON_EXECUTABLE=/usr/bin/python2 -DDESIRED_QT_VERSION=5 ..
`

This is on archlinux.

It also failed to generate doxygen docs, but lets ignore that for now. I disabled that and got to above output.

I have attached my [CMakeCache.txt](https://github.com/gnuradio/gnuradio/files/2877555/CMakeCache.txt). It does look like it did select python2 correctly.

drmpeg, 2019-02-19:
I can't reproduce this. Python 2 build works fine for me.

nickoe, 2019-02-19:
Ok, I think I found the root cause of this one. This was my mistake.

I was playing around with building for python3 and at some point I did a 2to3, just to see what would change. It removed the 
```
from __future__ import print_function
from __future__ import absolute_import
from __future__ import unicode_literals
```

Which are clearly needed for python2. Move on...   :)

closed 2019-02-19 by nickoe