gnuradio/gnuradio #2368: GRC: File -> Open Recent strips out underscores

nickoe, 2019-02-18:
It looks like the File -> Open Recent strips out underscores from the filenames.

Try to create a grc file named `/your_fancy_path/something_very_fancy_indeed.grc`, and save it. Then try to access the open recent menu and see it show `/yourfancypathsomethingveryfancyindeed.grc`.

3.8tech-preview-280-g3a533f78

nickoe, 2019-02-18:
By the way, I think it is only the view that is stripped, clicking the relevant entry in the open recent list do open the file properly with the underscores, where they are shown in the console output after "Loading: ".

haakov, 2019-02-19:
This happens because the underscores are treated as mnemonics. 

Does the following change fix the problem?

```diff
diff --git a/grc/gui/Bars.py b/grc/gui/Bars.py
index a42c13691..9f5049a89 100644
--- a/grc/gui/Bars.py
+++ b/grc/gui/Bars.py
@@ -152,7 +152,7 @@ class SubMenuHelper(object):
             files = Gio.Menu()
             for i, file_name in enumerate(recent_files):
                 target = "app.flowgraph.open_recent::{}".format(file_name)
-                files.append(file_name, target)
+                files.append(file_name.replace("_", "__"), target)
             menu.append_section(None, files)
             #clear = Gio.Menu()
             #clear.append("Clear recent files", "app.flowgraph.clear_recent")
```

noc0lour, 2019-02-19:
Hm, why is that though? Is this some Qt thing?

haakov, 2019-02-19:
Yeah you can try saving a file as `fg_x.grc`, closing it, opening the "open recent" drop-down menu next to the "open" button and pressing X. 

These underscores also enable you to use the Alt key, like Alt-F for File:
https://github.com/gnuradio/gnuradio/blob/3a533f78051afa4e33acada04fd6af6777f179f7/grc/gui/Bars.py#L67-L69

nickoe, 2019-02-19:
@haakov yea, that seems to fix it, both for python2 and python3 builds.

nickoe, 2019-02-21:
@haakov Are you going to make a pull request for this, or do you intend to fix this in a slightly other way which needs more testing?

haakov, 2019-02-21:
PR submitted: https://github.com/gnuradio/gnuradio/pull/2376

closed 2019-02-21 by mbr0wn