gnuradio/gnuradio #2369: GRC: Save As is broken, Save and Save Copy are OK

nickoe, 2019-02-18:
It looks like the "Save As"  menu does not work. I use the mount to select the menu.

When I try to "Save As" an existing grc it will not save. The normal Save for a modified or new grc filw works, and "Save Copy" works as well, even adding _copy to the basename.

3.8tech-preview-280-g3a533f78  with python3

aru31, 2019-02-19:
The problem with save_as was that `_handle_action()` was not called in case the grc file was already saved, which caused this problem. So, as a fix, I made a try_catch block for that particular action, thus solving the issue.

nickoe, 2019-02-19:
@aru31 your fix almost works, see https://github.com/gnuradio/gnuradio/pull/2373#issuecomment-465312494

nickoe, 2019-02-22:
@aru31 yay, your fix was merged. May I suggest you have add a "Fixes" line to your commit next time, then the issue the patch is to fix will be closed automagically by github.

If you don't know what I am talking about, please see:
https://help.github.com/en/articles/closing-issues-using-keywords

closed 2019-02-22 by nickoe

aru31, 2019-02-23:
Thank you @nickoe, will keep that in mind :)

noc0lour, 2019-02-24:
Adding the a `fixes #1234` as separate, last line is very encouraged. :)