gnuradio/gnuradio #2370: gnuradio 3.7.13.4, volk submodule 17277ed -  volk_32fc_x2_conjugate_dot_prod_32fc u_avx and a_avx bug.

mfogle, 2019-02-19:
The conj-dot-prod for the unaligned and aligned avx kernels appears to have a bug.  If the num_points in-param is not divisible by 4, a for-loop will perform the calculation on the remaining input and taps values.  The calculations in this for-loop are not accessing the input pointer (a_p) and taps pointer (b_p) incorrectly.  The attached patch file shows the issue and the fix.
[volk_32fc_x2_conjugate_dot_prod_32fc__17277ed.txt](https://github.com/gnuradio/gnuradio/files/2881277/volk_32fc_x2_conjugate_dot_prod_32fc__17277ed.txt)

noc0lour, 2019-02-20:
@mfogle can you submit an issue (or better, a pull request) in the https://github.com/gnuradio/volk repo?

closed 2019-02-20 by noc0lour

mfogle, 2019-02-20:
Mission accomplished.

_____________________
Michael Fogle
Signalscape, Inc.
mikef@signalscape.com
direct: (919)678-6581

--------------------------- This email and any files transmitted with it are confidential and intended solely for the use of Signalscape, Inc. and the addressed individual or entity. If you have received this email in error please delete it. Information in this email may be subject to the Privacy Act of 1974 and any unauthorized review, use, disclosure, or distribution is strictly prohibited. Any views or opinions presented in this email are solely those of the author and do not necessarily represent those of the company.

noc0lour, 2019-02-20:
Thanks a lot! :)