gnuradio/gnuradio #2374: GRC: Dark theme not working well

nickoe, 2019-02-19:
I use dark gtk theme and for me I can't see text in white input boxes as the text becomes white as well. I can see it in the inputboxes that are colored. Is there somewhere that text color is hardcoded?

[image]

I selected the rest of the test in the "Extra Dict." field, such that you can see that there are text input.

3.8tech-preview-280-g3a533f78

nickoe, 2019-02-24:
On a related note; the tabs in GRC to choose the active canvas has black text which does not work well on dark tabs.

chuangzhu, 2021-01-18:
Still doesn't work on KDE Plasma (5.20.5) when GTK theme set to `Breeze` and color scheme set to `Breeze Dark`. GRC version v3.8.2.0.

[image]

Here's my `~/.config/gtk-3.0/settings.ini`:

```ini
[Settings]
gtk-application-prefer-dark-theme=true
gtk-theme-name=Breeze
```

It seems that an unintentional mistake in #3017 resulted in `gtk-application-prefer-dark-theme` field not affecting the color:

```python
    prefer_dark = config.get(
        'Settings', Constants.GTK_INI_PREFER_DARK_KEY, fallback=None)
    if prefer_dark in ('1', 'yes', 'true', 'on'):
        theme_name = config.get(
            'Settings', Constants.GTK_INI_THEME_NAME_KEY, fallback=None)
-       return is_dark_theme(theme_name)
+       return True
    theme = subprocess.check_output(
        ["gsettings", "get", "org.gnome.desktop.interface", "gtk-theme"]
    ).decode('utf-8').strip().replace("'", "")
    return is_dark_theme(theme)
```

PR #4551 (2021-06-01): grc: fix dark theme on ParamWidgets
Fix `gtk-application-prefer-dark-theme` not affecting the color displayed.

For example, when you have a `~/.config/gtk-3.0/settings.ini` like this, the background color of the input box will still be white.

```
gtk-application-prefer-dark-theme=true
gtk-theme-name=Breeze
```

This also fixes #2374.
closed 2021-06-01 by mormj