gnuradio/gnuradio #2377: GnuradioConfig.cmake should contain list of available components, and Python version

mbr0wn, 2019-02-21:
When OOTs use GnuradioConfig.cmake (aka `find_package(Gnuradio)`), then there should be variables available that tell the OOT module whether or not modules are available (instead of just doing the required components thing).
In particular, right now, there's a but in gr-utils (somewhat related to this) that assumes ENABLE_GRC exists. That's a separate bug, because it shouldn't do that, but also, there's no good fix because OOTs have no way of knowing if GRC is available.

Also, the OOT needs to know which Python version was used to compile GNU Radio and default to that.

noc0lour, 2019-02-22:
Hm, this is a bit the other way around. Tools are allowed to depend on variables from the config (also we should prefix them with e.g. GR_). Also if you build a library depending on a certain GNU Radio component it must be sufficient to specify `REQUIRED COMPONENTS` and then the build process will fail if the component is not available. Otherwise without the required parameter you will get back a list of components that were found (subset of the specified list)'

bastibl, 2019-02-22:
At the moment modtool uses `ENABLE_GRC`, which is not propagated to OOTs, effectively disabling GRC bindings.
https://github.com/gnuradio/gnuradio/blob/master/gr-utils/python/modtool/templates/gr-newmod/CMakeLists.txt#L197-L199

So you would either introduce a GRC component or set it to `GR_ENABLE_GRC` in runtime?

noc0lour, 2019-02-22:
I think for that particular issue setting `GR_ENABLE_GRC` or similar would be best, since there is no real "GRC" target anyway.

noc0lour, 2019-03-08:
@marcusmueller is working on a solution for the python variables.

marcusmueller, 2019-03-08:
> At the moment modtool uses `ENABLE_GRC`, which is not propagated to OOTs, effectively disabling GRC bindings.
> https://github.com/gnuradio/gnuradio/blob/master/gr-utils/python/modtool/templates/gr-newmod/CMakeLists.txt#L197-L199
> 
> So you would either introduce a GRC component or set it to `GR_ENABLE_GRC` in GnuradioConfig?

In the meantime, I've merged a patch by drmpeg that includes the grc/ subdir in gr-newmod unconditionally – that's really suboptimal, but there's probably no harm done aside from modules which need to build hier blocks from GRC flow graphs at build time (which fail for less-than-clear reasons).

marcusmueller, 2019-04-21:
I believe this is fixed now, but @mbr0wn [could ideally](https://github.com/gnuradio/gnuradio/blob/master/cmake/Modules/GnuradioConfig.cmake.in#L57) verify (and if so, close), please?

closed 2019-05-16 by noc0lour