gnuradio/gnuradio #2383: Python String Formatting: Switch to f-strings

swap-nil7, 2019-03-01:
We should change the string formatting from `%s` or `str.format` to `f-strings` in the `Release 3.9`. It makes the code looks much cleaner.

jdemel, 2019-03-10:
This is a great Python feature. Though, the [Python f-string PEP](https://www.python.org/dev/peps/pep-0498/) says it was added as part of Python3.6. Thus, GR we'll only be able to use this feature as soon as we bump the minimum Python version to 3.6.

haakov, 2020-01-27:
So, now that the minimum Python version is 3.6, what exactly is the policy on f-strings? Should we modify existing code to use them or just start using them from now on?

devnulling, 2020-01-27:
I'd say new code should use f-strings.

I also think we should have a mega commit that switches all the existing strings to f-strings. This however though will make backporting fixes a bit more of a chore.

ritvikjain99, 2020-01-28:
@swap-nil7 I would like to take up this task.

swap-nil7, 2020-01-28:
@ritvikjain99 : Yeah, sure. To start with you can change the string formatting in the `gr-modtool`. Most of the changes will be there in the `core` directory of the tool.

mbr0wn, 2020-01-28:
@ritvikjain99 very cool! You might want to get your CLA process started so we don't have hold-ups down the line, if you haven't done so. Email bhilburn at gnuradio dot org to request a CLA form.

devnulling, 2020-02-14:
So for a follow up to this, we had more internal discussion and decided that:

- We wont have a mega commit at this time to convert to f-strings
- We will change the coding standards so new commits should use f-strings
- We will likely need to revisit this at some point in the future, but for now I'm going to close this issue.

closed 2020-02-14 by devnulling