gnuradio/gnuradio #2384: I have a problem with GUI

Yastro, 2019-03-01:
I have a problem with some of my GRC flow graphs that contain 'WX GUI FFT Sink'. Even with just 'RTL-SDR Source' connected to 'WX GUI FFT Sink' I get the following error message:

Warning: failed to XInitThreads()
linux; GNU C++ version 7.3.0; Boost_106501; UHD_003.010.003.000-0-unknown

Traceback (most recent call last):
File "/home/canyval/Desktop/fm_receiver.py", line 239, in <module>
main()
File "/home/canyval/Desktop/fm_receiver.py", line 233, in main
tb = top_block_cls()
File "/home/canyval/Desktop/fm_receiver.py", line 87, in __init__
win=window.hamming,
File "/usr/lib/python2.7/dist-packages/gnuradio/wxgui/fftsink_gl.py", line 92, in __init__
win=win,
File "/usr/lib/python2.7/dist-packages/gnuradio/fft/logpwrfft.py", line 65, in __init__
fft = self._fft_block[0](fft_size, True, fft_window)
File "/usr/lib/python2.7/dist-packages/gnuradio/fft/fft_swig.py", line 233, in make
return _fft_swig.fft_vfc_make(fft_size, forward, window, nthreads)
RuntimeError: std::exception

If I replace 'WX GUI FFT Sink' with 'WX GUI Scope Sink', the flow graph works.

bastibl, 2019-03-01:
I've only see this problem when GNU Radio and WX were compiled against different Python libraries. Any changes this might have happen (through some strange Pip/Virtualenv stuff)? How did you install GNU Radio?

Yastro, 2019-03-01:
Thanks your Comment.
I installed it using 'apt-get'.
But, the same problem also occurred when I installed it from the git

nickoe, 2019-03-07:
@Yastro What platform are you on? Can you check what wxgtk and wxpython packages you have installed?

marcusmueller, 2019-03-08:
> that contain 'WX GUI FFT Sink', QT Frequency Sink.

You can't have both at once.

nickoe, 2019-03-11:
Please close this issue.

closed 2019-03-11 by noc0lour