gnuradio/gnuradio #2386: gnuradio companion uses python2 when built cross with python3

balister, 2019-03-02:
Gnuradio built with OpenEmbedded uses python2 in gnuradio-companion, editing to python3 gives you a working grc.

nickoe, 2019-03-03:
How is this a problem with gnuradio? It sounds more like an issue with OpenEmbedded. Maybe I am just an outsider and does not quite follow what exactly you are trying to do with this issue.

drmpeg, 2019-03-04:
@nickoe There's plenty of if (CMAKE_CROSSCOMPILING) directives in the CMake files.

noc0lour, 2019-03-06:
@balister is referring to the Shebang-line carrying the wrong statement. This is due to `PYTHON_EXECUTABLE` being set to `python` which carries different meanings on the host & target machine. 
We might be able to support a separate cmake flag carrying additional information.

marcusmueller, 2019-03-08:
Yeah, let's face this: if you're crosscompiling, please set `-DPYTHON_EXECUTABLE=`.

I honestly fail to see how a build system is supposed to know when the target's `python` is different from the build machine's one; could you elaborate what you're wishing for, please, @balister ?

balister, 2019-03-10:
The build should default to python3, it looks like configure step finds
python3:

-- PYTHON_EXECUTABLE not set - using default python3
-- Use -DPYTHON_EXECUTABLE=/path/to/python2 to build for python2.
-- Found PythonInterp:
/home/balister/opensdr/ti-thud/build/tmp-glibc/work/armv7at2hf-neon-oe-linux-gnueabi/gnuradio/3.8.0+gitAUTOINC+61ebd24c4a-r0/recipe-sysroot-native/usr/bin/python3-native/python3
(found suitable version "3.7.2", minimum required is "3.4")
-- Found PythonLibs:

All the gnuradio modules are installed in the python3 site-packages.

Why does grc use python2 then? grc configure steps:

-- Python checking for python2 >= 2.7.6 or python3 >= 3.4.0 - found
-- Python checking for PyYAML >= 3.10 - not found
-- Python checking for mako >= 0.9.1 - found
-- Python checking for pygobject >= 2.28.6 - not found
-- Python checking for Gtk (GI) >= 3.10.8 - not found
-- Python checking for Cairo (GI) >= 1.0 - not found
-- Python checking for PangoCairo (GI) >= 1.0 - not found
-- Python checking for numpy - not found
-- 
-- Configuring gnuradio-companion support...
--   Dependency ENABLE_GNURADIO_RUNTIME = ON
--   Dependency ENABLE_PYTHON = ON
--   Enabling gnuradio-companion support.
--   Override with -DENABLE_GRC=ON/OFF

I'm trying to get around to reading cmake to see why grc has it's own
python check. Sadly, this is strictly a hobby project and gets spare
cycles when I can make them.

pingguokiller, 2019-03-26:
I got the same problem.  I want to use the gnuradio module in python3, but  failed and it's only available in python3.  Could someone help me!~

marcusmueller, 2019-04-21:
so, @balister, are you specifying `-DPYTHON_EXECUTABLE=python3` at CMake time? (btw, no, it should not default to `python3`. It should default to `python`.)

noc0lour, 2019-05-16:
```
else (PYTHON_EXECUTABLE)
    message(STATUS "PYTHON_EXECUTABLE not set - using default python3")
    message(STATUS "Use -DPYTHON_EXECUTABLE=/path/to/python2 to build for python2.")
    find_package(PythonInterp ${GR_PYTHON3_MIN_VERSION} REQUIRED)
```
@balister Python interpreter for CROSSCOMPILING is set to `python` regardless of the python chosen during build time.

balister, 2019-05-23:
I'm pretty sure I promised to change the hard coded interpreter to python3 when cross compiling. No sense investing lots of brain cells supporting python2 for this case. Unless someone want to give me a lot of money :)

closed 2020-08-30 by marcusmueller

noc0lour, 2020-08-31:
@balister can't you make sure the python on your embedded device is the right python? On all common other systems `python` represents the default python regardless of py2 or py3k.