gnuradio/gnuradio #2403: gr-modtool: Update gr_modtool info for the 3.8 release

swap-nil7, 2019-03-14:
We need to update the API version in the `gr_modtool info` for the `3.8 release`.

mormj, 2019-05-29:
Running gr_modtool info with the latest 3.8, I get the following:

Traceback (most recent call last):
  File "/share/gnuradio/grdev/bin/gr_modtool", line 30, in <module>
    cli()
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 1137, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.6/dist-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/share/gnuradio/grdev/lib/python3.6/dist-packages/gnuradio/modtool/cli/base.py", line 149, in wrapper
    return func(*args, **kwargs)
  File "/share/gnuradio/grdev/lib/python3.6/dist-packages/gnuradio/modtool/cli/info.py", line 42, in cli
    run(self)
  File "/share/gnuradio/grdev/lib/python3.6/dist-packages/gnuradio/modtool/cli/base.py", line 168, in run
    module.run()
  File "/share/gnuradio/grdev/lib/python3.6/dist-packages/gnuradio/modtool/core/info.py", line 50, in run
    raise ModToolException('{}' if self._python_readable else "No module found.")
gnuradio.modtool.core.base.ModToolException: No module found.

mormj, 2019-05-29:
I can dig into this if no one is actively debugging this issue

mormj, 2019-06-05:
@swap-nil7 - there is code already in there which does this but was in only one of two places where the version is being checked:

```
        if not os.path.isfile(os.path.join('cmake', 'Modules', 'FindCppUnit.cmake')):
            self.info['version'] = '38'
```
If I copy this to run() in core/info.py the version will pop up consistent with previous versions of the modtool info which is "post-3.8"

@marcusmueller - Is the idea with this issue just to continue with the modtool versioning as done in the past, i.e., "post-3.8"?  If so, this appears to be a simple fix.

swap-nil7, 2019-06-05:
Yes, I guess that's a correct fix.

mormj, 2019-06-06:
By the way, nice work @swap-nil7 on the modtool refresh!

PR #2533 (2019-06-15): modtool: add 3.8 version for info command
Add version information detection to the part of the code being called by gr_modtool info

fixes #2403
commit 25b01217 (2019-06-15): modtool: add 3.8 version for info command

Code was there to give the correct printout, just in another place
Copy the logic for finding the post-3.8 version information
to the code that is executed by the gr_modtool info command

fixes #2403
closed 2019-06-15 by mbr0wn