gnuradio/gnuradio #2405: bug in buffer_reader::get_tags_in_range

hcab14, 2019-03-15:
I got a segmentation fault in `buffer_reader::get_tags_in_range`: it occurs
for the following parameters:
```
abs_start = 3449
abs_end = 3961
d_attr_delay =  3718
item_time = 3718
```
in L371, see below.

With these inputs `itr = d_buffer->get_tags_end()` and `itr != itr_end`.

Note that for the inputs above, `abs_start < d_attr_delay` and therefore `abs_start - d_attr_delay` wraps around (`uint64_t`).

I see two options to fix this
1. replace `abs_start - d_attr_delay` by `abs_start - std::min(abs_start,uint64_t(d_attr_delay))`, and analogously for `abs_end - d_attr_delay`
2. in the while loop condition replace `itr != itr_end` with `itr != itr_end && itr != d_buffer->get_tags_end()`

Option 1 seems to be the correct one in my opinion.

https://github.com/gnuradio/gnuradio/blob/d5e0b4b82aa2f08030d391cc878698295e525b4a/gnuradio-runtime/lib/buffer.cc#L355-L382

marcusmueller, 2019-04-21:
hey, it's hard to reproduce that; can you give us something to play with?

willcode, 2021-03-22:
Fixed in 3.8 https://github.com/gnuradio/gnuradio/pull/3040. Fixed in later versions by https://github.com/gnuradio/gnuradio/pull/2806. Closing.

closed 2021-03-22 by willcode