gnuradio/gnuradio #2407: heap-buffer-overflow in gr-qtgui/lib/freq_sink_c_impl.cc

hcab14, 2019-03-16:
https://github.com/gnuradio/gnuradio/blob/d5e0b4b82aa2f08030d391cc878698295e525b4a/gr-qtgui/lib/freq_sink_c_impl.cc#L509

I guess it should be `sizeof(float)*(d_tmpbuflen)` instead of `sizeof(float)*(d_tmpbuflen + 1)`? Can anyone confirm?

Below is the trace from address sanitizer for GNURadio 3.7.12 but the offending line of code has not changed since.

The heap-buffer-overflow can be triggered with this simple flow graph: [bug.grc.txt](https://github.com/gnuradio/gnuradio/files/2973819/bug.grc.txt)

```
=================================================================
==22939==ERROR: AddressSanitizer: heap-buffer-overflow on address 0x621000d59500 at pc 0x7f455ac7077a bp 0x7f452eee78b0 sp 0x7f452eee7058
WRITE of size 2052 at 0x621000d59500 thread T6 (freq_sink_c1)
    #0 0x7f455ac70779  (/usr/lib/gcc/x86_64-linux-gnu/7/libasan.so+0x79779)
    #1 0x7f45344fc7b0 in memcpy /usr/include/x86_64-linux-gnu/bits/string_fortified.h:34
    #2 0x7f45344fc7b0 in gr::qtgui::freq_sink_c_impl::fft(float*, std::complex<float> const*, int) /home/hcab14/gnuradio/gr-qtgui/lib/freq_sink_c_impl.cc:500
    #3 0x7f453450040d in gr::qtgui::freq_sink_c_impl::work(int, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&) /home/hcab14/gnuradio/gr-qtgui/lib/freq_sink_c_impl.cc:701
    #4 0x7f4549d6a760 in gr::sync_block::general_work(int, std::vector<int, std::allocator<int> >&, std::vector<void const*, std::allocator<void const*> >&, std::vector<void*, std::allocator<void*> >&) /home/hcab14/gnuradio/gnuradio-runtime/lib/sync_block.cc:66
    #5 0x7f4549c46ffc in gr::block_executor::run_one_iteration() /home/hcab14/gnuradio/gnuradio-runtime/lib/block_executor.cc:500
    #6 0x7f4549d8da04 in gr::tpb_thread_body::tpb_thread_body(boost::shared_ptr<gr::block>, boost::shared_ptr<boost::barrier>, int) /home/hcab14/gnuradio/gnuradio-runtime/lib/tpb_thread_body.cc:123
    #7 0x7f4549d5863a in gr::tpb_container::operator()() /home/hcab14/gnuradio/gnuradio-runtime/lib/scheduler_tpb.cc:46
    #8 0x7f4549d5863a in gr::thread::thread_body_wrapper<gr::tpb_container>::operator()() /home/hcab14/gnuradio/gnuradio-runtime/include/gnuradio/thread/thread_body_wrapper.h:51
    #9 0x7f4549d5863a in boost::detail::function::void_function_obj_invoker0<gr::thread::thread_body_wrapper<gr::tpb_container>, void>::invoke(boost::detail::function::function_buffer&) /usr/include/boost/function/function_template.hpp:159
    #10 0x7f4549bdf914 in boost::function0<void>::operator()() const /usr/include/boost/function/function_template.hpp:759
    #11 0x7f4549bdf914 in boost::detail::thread_data<boost::function0<void> >::run() /usr/include/boost/thread/detail/thread.hpp:116
    #12 0x7f4548f93bcc  (/usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1+0x11bcc)
    #13 0x7f455a5ee6da in start_thread (/lib/x86_64-linux-gnu/libpthread.so.0+0x76da)
    #14 0x7f455a92788e in __clone (/lib/x86_64-linux-gnu/libc.so.6+0x12188e)

0x621000d59500 is located 0 bytes to the right of 4096-byte region [0x621000d58500,0x621000d59500)
allocated by thread T0 here:
    #0 0x7f455acd67a0 in posix_memalign (/usr/lib/gcc/x86_64-linux-gnu/7/libasan.so+0xdf7a0)
    #1 0x7f454920ea0b in volk_malloc /home/hcab14/gnuradio/volk/lib/volk_malloc.c:49

Thread T6 (freq_sink_c1) created by T0 here:
    #0 0x7f455ac2ed2f in __interceptor_pthread_create (/usr/lib/gcc/x86_64-linux-gnu/7/libasan.so+0x37d2f)
    #1 0x7f4548f92b09 in boost::thread::start_thread_noexcept() (/usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1+0x10b09)

SUMMARY: AddressSanitizer: heap-buffer-overflow (/usr/lib/gcc/x86_64-linux-gnu/7/libasan.so+0x79779) 
Shadow bytes around the buggy address:
  0x0c42801a3250: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c42801a3260: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c42801a3270: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c42801a3280: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
  0x0c42801a3290: 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
=>0x0c42801a32a0:[fa]fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c42801a32b0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c42801a32c0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c42801a32d0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c42801a32e0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
  0x0c42801a32f0: fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa fa
Shadow byte legend (one shadow byte represents 8 application bytes):
  Addressable:           00
  Partially addressable: 01 02 03 04 05 06 07 
  Heap left redzone:       fa
  Freed heap region:       fd
  Stack left redzone:      f1
  Stack mid redzone:       f2
  Stack right redzone:     f3
  Stack after return:      f5
  Stack use after scope:   f8
  Global redzone:          f9
  Global init order:       f6
  Poisoned by user:        f7
  Container overflow:      fc
  Array cookie:            ac
  Intra object redzone:    bb
  ASan internal:           fe
  Left alloca redzone:     ca
  Right alloca redzone:    cb
==22939==ABORTING
```

noc0lour, 2019-03-16:
Hm looking at the code this isn't the problem since the tmpbuf is created with the appropriate length:
```
      d_tmpbuf = (float*)volk_malloc(sizeof(float)*(d_tmpbuflen + 1),
                                     volk_get_alignment());
```

Looking at the line numbers it looks like something is fishy with: 
```
	memcpy(d_fft->get_inbuf(), data_in, sizeof(gr_complex)*size);
```

hcab14, 2019-03-16:
> Looking at the line numbers it looks like something is fishy with:
> 	`memcpy(d_fft->get_inbuf(), data_in, sizeof(gr_complex)*size);`

the heap buffer overflow does not happen ion this line.

https://github.com/gnuradio/gnuradio/blob/d5e0b4b82aa2f08030d391cc878698295e525b4a/gr-qtgui/lib/freq_sink_c_impl.cc#L106

https://github.com/gnuradio/gnuradio/blob/d5e0b4b82aa2f08030d391cc878698295e525b4a/gr-qtgui/lib/freq_sink_c_impl.cc#L107-L108

https://github.com/gnuradio/gnuradio/blob/d5e0b4b82aa2f08030d391cc878698295e525b4a/gr-qtgui/lib/freq_sink_c_impl.cc#L102-L103

https://github.com/gnuradio/gnuradio/blob/d5e0b4b82aa2f08030d391cc878698295e525b4a/gr-qtgui/lib/freq_sink_c_impl.cc#L723

https://github.com/gnuradio/gnuradio/blob/d5e0b4b82aa2f08030d391cc878698295e525b4a/gr-qtgui/lib/freq_sink_c_impl.cc#L490-L491

https://github.com/gnuradio/gnuradio/blob/d5e0b4b82aa2f08030d391cc878698295e525b4a/gr-qtgui/lib/freq_sink_c_impl.cc#L509

so here `data_out = d_fbuf` and copying `d_tmpbuflen + 1` items starting at `d_tmpbuflen` causes the heap buffer overflow.

mormj, 2019-05-15:
Compiling in ASAN to gr-qtgui, issue was easily repeatable with a simple flowgraph:

[image]

Saw same heapdump as above pointing to line 723 - the call to fft().  Source level debugging confirmed the overflow happening at line 509 (as debugged by @hcab14 .  Removed the +1 and problem goes away, and it makes sense.  Will submit pull request with the +1 removed.

mormj, 2019-06-18:
This issue can be closed as it was fixed in @hcab14 's commits and merged as referenced above.  Verified the merged changes fix this issue specifically

closed 2019-06-18 by mormj