gnuradio/gnuradio #2421: Recompile fails after `gr_modtool add`

jdemel, 2019-03-24:
After an OOT is built and a new block is added subsequently, CMake seems to be missing Thrift again. This error may be fixed with a new build folder. 

`gr_modtool newmod testmod`
`cd testmod`
`gr_modtool add testblock [sync block, python or cpp, no arguments]`
`mkdir build`
`cd build`
`cmake ..`
`make`
`cd ..`
`gr_modtool add anotherblock [sync block, python or cpp, no arguments]`
`cd build`
`make`

> CMake Error at lib/CMakeLists.txt:37 (add_library):
>   Target "gnuradio-testmod" links to target "Thrift::thrift" but the target was
>   not found.  Perhaps a find_package() call is missing for an IMPORTED
>   target, or an ALIAS target is missing?
> 
> CMake Error at [prefix]/lib/cmake/gnuradio/UseSWIG.cmake:573 (add_library):
>   Target "testmod_swig" links to target "Thrift::thrift" but the target was not
>   found.  Perhaps a find_package() call is missing for an IMPORTED target, or
>   an ALIAS target is missing?
> Call Stack (most recent call first):
>   [prefix]/lib/cmake/gnuradio/GrSwig.cmake:133 (swig_add_library)
>   swig/CMakeLists.txt:51 (GR_SWIG_MAKE)
> 
> CMake Error at lib/CMakeLists.txt:37 (add_library):
>   Target "gnuradio-testmod" links to target "Thrift::thrift" but the target was
>   not found.  Perhaps a find_package() call is missing for an IMPORTED
>   target, or an ALIAS target is missing?

`make` notices that some `CMakeLists.txt` files were updated and reruns `cmake`. Unfortunately, this seems to fail.

noc0lour, 2019-03-24:
Weird issue. Needs investigation what happens there exactly.

jdemel, 2019-03-24:
This issue arises after the configuration stage and only during the cmake generation stage.

jdemel, 2019-04-01:
In case @marcusmueller proposal to remove ctrlport/Thrift is passed, this issue would be solved as well.

noc0lour, 2019-05-16:
I could reproduce this issue locally, but after several iterations of wiping my build dir & cmake & ninja building succeeded. Something is definitely not right with thrift.

NeuroForLunch, 2021-06-24:
Stale issue. No version information. Should be closed.

jdemel, 2021-06-25:
> Stale issue. No version information. Should be closed.

I agree. Closing it now.

closed 2021-06-25 by jdemel