gnuradio/gnuradio #2427: Instantiating a gr::fft::fft_complex causes an exception.

johnwstanford, 2019-03-30:
I found this while trying to run GNSS-SDR, but I get an exception when I try to instantiate a complex FFT.  Here's a simple example independent of GNSS-SDR:

Source code:
```
#include <iostream>

#include <boost/thread.hpp>

#include <gnuradio/fft/api.h>
#include <gnuradio/fft/fft.h>
#include <gnuradio/gr_complex.h>

int main() {

	auto fft = new gr::fft::fft_complex(2000, false);

	std::cout << "Success" << std::endl;

}
```

Result:
```
john@deathStar:~/Desktop/gr_test$ g++ ./main.cpp -lboost_system -lgnuradio-fft -o main
john@deathStar:~/Desktop/gr_test$ ./main
terminate called after throwing an instance of 'std::exception'
  what():  std::exception
Aborted (core dumped)
```

noc0lour, 2019-03-30:
Hm, can you show me `ldd main`?  Exactly the same C++ file works on my machine.

johnwstanford, 2019-03-31:
Here's what I get on my computer:

```
john@deathStar:~/Desktop/gr_test$ ldd ./main
	linux-vdso.so.1 (0x00007ffff41a5000)
	libboost_system.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_system.so.1.65.1 (0x00007f99ae733000)
	libgnuradio-fft.so.3.7.11 => /usr/lib/x86_64-linux-gnu/libgnuradio-fft.so.3.7.11 (0x00007f99ae4ff000)
	libstdc++.so.6 => /usr/lib/x86_64-linux-gnu/libstdc++.so.6 (0x00007f99ae176000)
	libgcc_s.so.1 => /lib/x86_64-linux-gnu/libgcc_s.so.1 (0x00007f99adf5e000)
	libc.so.6 => /lib/x86_64-linux-gnu/libc.so.6 (0x00007f99adb6d000)
	libpthread.so.0 => /lib/x86_64-linux-gnu/libpthread.so.0 (0x00007f99ad94e000)
	libgnuradio-runtime.so.3.7.11 => /usr/lib/x86_64-linux-gnu/libgnuradio-runtime.so.3.7.11 (0x00007f99ad65e000)
	libvolk.so.1.3 => /usr/lib/x86_64-linux-gnu/libvolk.so.1.3 (0x00007f99ad296000)
	libboost_filesystem.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_filesystem.so.1.65.1 (0x00007f99ad07c000)
	libboost_thread.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.65.1 (0x00007f99ace57000)
	libfftw3f.so.3 => /usr/lib/x86_64-linux-gnu/libfftw3f.so.3 (0x00007f99aca4a000)
	libfftw3f_threads.so.3 => /usr/lib/x86_64-linux-gnu/libfftw3f_threads.so.3 (0x00007f99ac843000)
	libgnuradio-pmt.so.3.7.11 => /usr/lib/x86_64-linux-gnu/libgnuradio-pmt.so.3.7.11 (0x00007f99ac5ee000)
	libm.so.6 => /lib/x86_64-linux-gnu/libm.so.6 (0x00007f99ac250000)
	/lib64/ld-linux-x86-64.so.2 (0x00007f99aeb44000)
	libboost_program_options.so.1.65.1 => /usr/lib/x86_64-linux-gnu/libboost_program_options.so.1.65.1 (0x00007f99abfcf000)
	liblog4cpp.so.5 => /usr/lib/x86_64-linux-gnu/liblog4cpp.so.5 (0x00007f99abd92000)
	librt.so.1 => /lib/x86_64-linux-gnu/librt.so.1 (0x00007f99abb8a000)
	liborc-0.4.so.0 => /usr/lib/x86_64-linux-gnu/liborc-0.4.so.0 (0x00007f99ab90e000)
john@deathStar:~/Desktop/gr_test$ 

```

noc0lour, 2019-03-31:
Hm, everything looks okay. So I have seen similar issues (I think) that were related with fftw being linked with and without `fftw_threads` I think @drmpeg did see those issues as well?

johnwstanford, 2019-03-31:
I re-imaged the partition I was running this on with a fresh copy of Ubuntu 18.02, re-installed all the dependencies, and now it runs fine.  The output of `ldd ./main` is the same too (at least the same shared library names).  Maybe one of the library files was corrupted somehow.  I guess I could have compared hashes for those library files, but I didn't think of that before I re-imaged the partition :/  Should I go ahead and close the issue?

noc0lour, 2019-04-01:
I close it and if you see the issue again you can reopen it.

closed 2019-04-01 by noc0lour