gnuradio/gnuradio #2429: gr_modtool generates bad cmake include

drmpeg, 2019-03-31:
When trying to include another already installed OOT into a current OOT, gr_modtool creates a CMake file that tries to include a non-existent file.

In this example, gr-iqlevels wants to include gr-paint. In the top level CMakeLists,txt, this find_package() is added:

find_package(paint)

The find_package works correctly, but paintConfig.cmake can't find paintTarget.cmake since it doesn't exist.

Here's the CMake dump:
```
re@beavis:~/xfer/gr-iqlevels/build$ cmake -DCMAKE_INSTALL_PREFIX=/opt/gnuradio-3.7.12git ../
-- The CXX compiler identification is GNU 5.4.0
-- The C compiler identification is GNU 5.4.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Build type not specified: defaulting to release.
-- Found LOG4CPP: /usr/lib/liblog4cpp.so
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Checking for module 'gmp'
--   No package 'gmp' found
-- Found GMP: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Checking for module 'mpir >= 3.0'
--   No package 'mpir' found
-- Could NOT find MPIR (missing: MPIRXX_LIBRARY MPIR_LIBRARY MPIR_INCLUDE_DIR) 
-- Found MPLIB: /usr/lib/x86_64-linux-gnu/libgmpxx.so  
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   date_time
--   program_options
--   filesystem
--   system
--   regex
--   thread
--   unit_test_framework
--   chrono
--   atomic
-- User set python executable /usr/bin/python3
-- Found PythonInterp: /usr/bin/python3 (found version "3.5.2") 
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.5m.so (found suitable exact version "3.5.2") 
-- Found Git: /usr/bin/git  
-- Checking for module 'paint'
--   No package 'paint' found
CMake Error at /opt/gnuradio-3.7.12git/lib/cmake/paint/paintConfig.cmake:27 (include):
  include could not find load file:

    /opt/gnuradio-3.7.12git/lib/cmake/paint/paintTarget.cmake
Call Stack (most recent call first):
  CMakeLists.txt:94 (find_package)

-- Found PAINT: /opt/gnuradio-3.7.12git/lib/libgnuradio-paint.so  
-- Paint Found
-- Found Doxygen: /usr/bin/doxygen (found version "1.8.11") found components:  doxygen dot 
-- Using install prefix: /opt/gnuradio-3.7.12git
-- Building for version: v1.0-compat-xxx-xunknown / 1.0.0git
-- No C++ unit tests... skipping
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.8.
-- Found SWIG: /usr/bin/swig3.0  
-- Found PythonLibs: /usr/lib/x86_64-linux-gnu/libpython3.5m.so (found version "3.5.2") 
-- Configuring incomplete, errors occurred!
See also "/home/re/xfer/gr-iqlevels/build/CMakeFiles/CMakeOutput.log".
See also "/home/re/xfer/gr-iqlevels/build/CMakeFiles/CMakeError.log".
re@beavis:~/xfer/gr-iqlevels/build$ 
```

marcusmueller, 2019-04-21:
I'd like to ask @noc0lour to have a look at this after May 2.

noc0lour, 2019-04-23:
Including an OOT in yet another cmake project is currently not working correctly, so definitely a bug.

noc0lour, 2019-05-16:
Hm, is this with 3.8 code? your first line states a GNU Radio 3.7.12 prefix?

drmpeg, 2019-05-17:
Yes, it's 3.8. I neglected to change the install prefix to something more meaningful.

noc0lour, 2019-05-17:
Since then the modtool template has been updated. Basically you need to copy `gr-utils/python/modtool/templates/gr-newmod/cmake/Modules/targetConfig.cmake.in` to the `cmake/Modules/` of gr-paint. If you use the most recent version of GNU Radio master GR_LIBRARY_FOO should do the trick.

noc0lour, 2019-07-17:
This needs a final fix before the release.

dkozel, 2021-09-27:
@drmpeg @noc0lour This seems likely to have been fixed by now?

drmpeg, 2021-09-27:
Probably not. But including another OOT in your OOT seems to be pretty much never used, so this issue is very obscure.

drmpeg, 2021-10-08:
@dkozel A conversation with Maitland on IRC showed me the error of my ways. I was using the wrong ```find_package()``` invocation. Instead of ```find_package(paint)```, it should be find_package(gnuradio-paint).

I was able to build, link and run an OOT that included gr-paint.

drmpeg, 2021-10-08:
Just in case someone finds this in the future, here are the changes to implement this (made to gr-iqlevels as a test). Just to show it can be linked, I'm calling ```paint_bc::make()```. In addition, you need a CMake path to gr-paint. ```export CMAKE_PREFIX_PATH=<gnuradio_install_dir>/lib/cmake/paint```
```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index f028418..e3f1ac1 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -75,6 +75,8 @@ ENDIF()
 ########################################################################
 include(FindPkgConfig)
 find_package(Gnuradio "3.9" REQUIRED)
+find_package(gnuradio-paint)
+find_package(Gnuradio COMPONENTS fft)
 include(GrVersion)
 
 include(GrPlatform) #define LIB_SUFFIX
diff --git a/lib/CMakeLists.txt b/lib/CMakeLists.txt
index 5cbe07f..3728873 100644
--- a/lib/CMakeLists.txt
+++ b/lib/CMakeLists.txt
@@ -22,7 +22,7 @@ if(NOT iqlevels_sources)
 endif(NOT iqlevels_sources)
 
 add_library(gnuradio-iqlevels SHARED ${iqlevels_sources})
-target_link_libraries(gnuradio-iqlevels gnuradio::gnuradio-runtime)
+target_link_libraries(gnuradio-iqlevels gnuradio::gnuradio-runtime gnuradio::gnuradio-fft gnuradio::gnuradio-paint)
 target_include_directories(gnuradio-iqlevels
     PUBLIC $<BUILD_INTERFACE:${CMAKE_CURRENT_SOURCE_DIR}/../include>
     PUBLIC $<INSTALL_INTERFACE:include>
diff --git a/lib/iqlevels_impl.cc b/lib/iqlevels_impl.cc
index d50de29..dff7d3b 100644
--- a/lib/iqlevels_impl.cc
+++ b/lib/iqlevels_impl.cc
@@ -25,6 +25,7 @@
 #include <gnuradio/io_signature.h>
 #include "iqlevels_impl.h"
 #include <stdio.h>
+#include <paint/paint_bc.h>
 
 namespace gr {
   namespace iqlevels {
@@ -44,6 +45,7 @@ namespace gr {
               gr::io_signature::make(1, 1, sizeof(gr_complex)),
               gr::io_signature::make(0, 0, 0))
     {
+      gr::paint::paint_bc::sptr paint_enc;
       real_positive = 0.0;
       real_negative = 0.0;
       imag_positive = 0.0;
@@ -60,6 +62,7 @@ namespace gr {
       delay_count = delay;
       show_levels = showlevels;
       set_output_multiple(CHUNKS);
+      paint_enc = gr::paint::paint_bc::make(1920, 5, gr::paint::EQUALIZATION_OFF, gr::paint::INTERNAL, 1);
     }
 
     /*
```

ckoehler, 2021-11-03:
I am not sure this is fixed in master. `gr_modtool newmod foo` defines `set(GR_CMAKE_DIR        ${CMAKE_MODULES_DIR}/foo)` and installs an included `fooConfig.cmake` there:

```
install(FILES cmake/Modules/fooConfig.cmake
    DESTINATION ${CMAKE_MODULES_DIR}/foo
)
```

So when I install `foo` and want to include it in another module, it looks like I should just do `find_package(foo)`, which will look in `<somewhere>/lib/cmake/foo/` and find the config there.

However, `fooConfig.cmake` includes a non-existent file, causing CMake to fail: `include("${CMAKE_CURRENT_LIST_DIR}/fooTarget.cmake")`

The way it actually works is the `GR_LIBRARY_FOO` macro is called with the target name, which is by default `gnuradio-foo`. The macro creates a CMake package config for that target called `gnuradio-fooConfig.cmake` and the export targets in file `gnuradio-fooTargets.cmake`, both installed into `GR_CMAKE_DIR` (see definition above: no `gnuradio-` prefix). 

Assuming any `find_package` call should be done with the full `gnuradio-foo` name, the fix here would be to remove the included `fooConfig.cmake` file and its install directive, and change `GR_CMAKE_DIR` to be `${CMAKE_MODULES_DIR}/gnuradio-foo`.

Thoughts? Is that the intended way to `find_package` a GR OOT lib?

drmpeg, 2021-11-03:
What needs to be fixed? It works fine if you do ```find_package(gnuradio-foo)```, doesn't it?

ckoehler, 2021-11-03:
No, because `<somewhere>/lib/cmake/gnuradio-foo/` doesn't exist, so CMake just aborts. The fix I mentioned above would make sure that it does exist so CMake can find it.

drmpeg, 2021-11-03:
It works for me (as described above).

ckoehler, 2021-11-03:
But you have to manipulate the `CMAKE_PREFIX_PATH` manually. It won't work without that, and it's just a hack. The CMake package config created by default is broken, or it would just work.

drmpeg, 2021-11-03:
Oh yeah, I remember now. I agree that setting the ```CMAKE_PREFIX_PATH``` is an unnecessary complication. You should go ahead with your PR for sure.

PR #5301 (2021-11-12): modtool: fix CMake export config
# Pull Request Details

## Description

Adjusts the directory where CMake installs exported targets, so OOT modules can be found via `find_package`. This functionality is already present in the `GR_LIBRARY_FOO` macro, the only thing preventing it from working was a wrong default `GR_CMAKE_DIR` path. This is fixed.

This also removes the old CMake config and its install directive.

## Related Issue

Fixes #2429

## Which blocks/areas does this affect?

modtool

##