gnuradio/gnuradio #2430: Bump minimum Python3 version to Python 3.5

noc0lour, 2019-03-31:
`qa_bin_statistics` only works for Python 3.5 and greater due to the line 
`self.data = struct.unpack(b'%df' % self.vlen, msg.to_string())`.

It's not really affecting anything aside from the QA tests, but there is nothing bad bumping the version to 3.5 or even 3.6. We are living in 2019 after all.

jdemel, 2019-04-01:
Ubuntu 16.04 comes with Python 3.5. Thus, I'd argue that 3.5 should be supported. Also, Debian stretch (stable) comes with 3.5. With that in mind, it would be a good idea to bump the Python 3 minimum version to 3.5. What about Fedora etc.?

noc0lour, 2019-04-01:
Definitely. Current fedoras have a new enough python3 same for centos with epel-testing.

michaelld, 2019-04-01:
Py35 min will be fine for OSX users.

marcusmueller, 2019-04-21:
I'd be all for bumping straight to 3.6, if @michaelld is fine with that. 3.8 has no obligation to be Ubuntu16.04 compatible; 16.04 is in "maintenance updates only" mode, which any other distro would call "unsupported".

michaelld, 2019-04-21:
Py36 is fine. I can build and run Py36 back on OSX from like 10 years ago, probably older. That's plenty of backward compatibility if you ask me!

marcusmueller, 2019-04-21:
Well! Since EPEL on CentOS / RedHat 7 supports Py3.6, and so does Ubuntu 18.04, I'd say we move to 3.6

marcusmueller, 2019-04-21:
specifically, Ubuntu 18.04 seems to have the oldest Python3 (3.6.5) of all relevant OSes, so let's use that version.

marcusmueller, 2019-04-21:
Email to the mailing list went out. Waiting for feedback till 2019-05-01

Juanlu001, 2019-04-22:
The current Debian stable, Stretch, ships Python 3.5:

https://packages.debian.org/stable/python3

Also, according to the JetBrains survey, Python 3.5 is used by 30 % of the users:

https://www.jetbrains.com/research/python-developers-survey-2018/#python-3-adoption

Which more or less agrees with the download stats, 3.5 being second in Python 3 versions:

https://pypistats.org/packages/pip

I admit Stretch will be oldstable in a few months and that Python 3.5 is not exactly "bleeding edge" (will be EOL in ~September 2020) but given that there is no other gnuradio release with Python 3 support, that Python 3.5 was the first version to drive major Python 3 adoption and that not all the users can live in the bleeding edge, it would be great to have 3.5 support :blush: My 2 cents!

Juanlu001, 2019-04-22:
> I can build and run Py36 back on OSX from like 10 years ago, probably older.

How come? :) https://www.python.org/dev/peps/pep-0494/#release-schedule

marcusmueller, 2019-04-22:
@Juanlu001 thanks for the insight! The 30% would be a stronger argument if the JetBrains study wasn't focused on web developers – both manifested in their numbers on what Python is used for and what the other languages of the projects are.

The good news is that as far as I see, web development and devops is very heavily done with virtualenvs/anaconda envs etc, so that these folks already know how to get a prefix with a newer Python installed. The pypi/pip stats support that hypothesis: Of the python users that never download pip (i.e. those who simply use the python their distro offers, and the pip that goes along with it), the majority by now will have access to systems at least as recent as Ubuntu 18.04, so they definitely got Python 3.6 going. My hypothesis is that pip downloads are typical for these using pip to get dependencies for specific virtualenv projects, which might be running on conservatively administrated servers. That's not what we need to be compatible with!

The debian stretch argument is a *very* valid one – I definitely want GNU Radio 3.8 to be supported by Debian; it's my favourite system for people to use GNU Radio on. I'll get in touch with our debian maintainer asking how realistic it is that debian buster is out by then. Also, I think chances are that debian stretch, being stable, simply would never ship GNU Radio 3.8, so we don't even have to try to be compatible with it. Again, let me discuss this with our debian maintainer. I won't allow myself to rush this decision (which is why I've put a long [latency](https://github.com/gnuradio/gnuradio/issues/2430#issuecomment-485281955) on this issue). 

You'll notice that I seem, compared to how the project used to be run, are trying to be pretty aggressive with dependency versioning. My rationale is this:

GNU Radio 3.8's dependencies are what we're stuck for quite a long time. A freshly developed version doesn't have to support legacy systems – it needs to work with the ones widely in use at its release! People trying to run bleeding edge software on legacy systems might need to jump a few hoops, but I honestly think these hoops are jumpable, and I'll do my developers a favor if I'm a bit progressive with what versions of dependencies they can rely on having. Especially my Pythonistas tend to be pushing me towards modern Python and Python packages (looking at @skoslowski there, for example).

michaelld, 2019-04-22:
> > I can build and run Py36 back on OSX from like 10 years ago, probably older.
> 
> How come? :) https://www.python.org/dev/peps/pep-0494/#release-schedule

I'm assuming your note is with regard to the age of Py36 (3.6.0 released 2016-12-23, or about 2.5 years ago) versus my note (from like 10 years ago) ... yes? You & I both know that in general newer projects can be built for older OSs, and Py36 is not an exception here. One can use MacPorts, Fink, Homebrew, or some homemade "from source" way ... it's all about the same. Apple of course doesn't necessarily provide Py36 with their default OS install, but it's not difficult to install Py36 & have it work as well as if it were installed with the OS. Anyway, I don't see bumping to Py36 as a limitation on OSX. ... I have it running nicely on my OSX 10.5 PPC iMac which dates back to maybe 2005 ... as well as more modern OSX 10.6 to 10.14 Intel on various MacBook laptops from 2009 to 2017. Python is amazingly compatible! I can't say that for lots of projects that make significant use of OSX APIs (e.g., GLFW, FLTK, Qt4/5) ... those end up being difficult to support across various OSX versions, since Apple changes and deprecates APIs regularly.

Juanlu001, 2019-04-22:
Thanks both for your insightful responses!

marcusmueller, 2019-04-23:
@Juanlu001, no thank you for raising these points! It's greatly appreciated.

I'll quickly cite Maitland, our debian maintainer of trust, whom I asked:

> > That was under the assumption that 
> > a) debian buster becomes stable before we release GR 3.8 and 
> > b) we get GNU Radio 3.8 into stable, anyways (current testing does
> > already ship Python >=3.6, doesn't it?)

> Debian 10 Buster will ship with gnuradio 3.7.13.4 and python3 3.7.2-1.
> Those are the versions in testing now during the release freeze.

So the thing that will ship with the initial release of buster is GR 3.7.13.
Since they are in release freeze, I don't feel there will be many people trying to use 3.8 or develop for 3.9 on stretch.

> > So, does it even make sense to target debian stretch with GNU Radio
> > 3.8?
> No.
> […] by policy, stretch backports will only get up to gnuradio 3.7.13.4.

and

> > I'd be a bad maintainer if my project broke
> > under debian stable, and I couldn't stand that.

> Go ahead and bump the minimum Python3 dependency version for the 3.8
> release. We want gnuradio in Debian 11 Bullseye to totally rock.

So, that's a "go" on Python 3.6 from the debian side.

jdemel, 2019-04-23:
If we bump the minimum Python version to 3.6, we'd be able to address #2383 right away.

marcusmueller, 2019-04-26:
@jdemel no, since for 3.8 we still support python 2.7

michaelld, 2019-04-27:
@marcusmueller so if we're still supporting Py27 in GR38, then are we targeting moving away from Py27 come GR39? We tend to break things between "minor" releases (3.6 -> 3.7 did; and now 3.7 -> 3.8 will again), so this timing would make sense to update dependency requirements to reasonably modern standards -- assuming we need to and it makes sense to do so (at least IMHO).

noc0lour, 2019-04-27:
@michaelld yes, the day 3.8 is released I will kill py27 support on master personally. (at least on my branch).

mbr0wn, 2019-06-06:
Does 3.6 have something important? Can we just stick with 3.5 for now? I don't think GNU Radio 3.9 will be that far out compared to 3.8, and thus we won't be stuck with these deps forever.

mbr0wn, 2019-07-11:
Decision was made on 11-July-2019: We'll go to 3.6. Rationale:
- We need to bump the version anyway, because 3.4 (the current min version) doesn't work
- Might as well go straight to 3.6
- Out of the popular OSes, only Ubuntu 16.04 doesn't have 3.6 (but note that installing custom Python versions is easier than other stuff, via pip etc.)
- It gives us format strings

PR #2608 (2019-07-13): cmake: deps: Bump min Python version to 3.6
Closes #2430
closed 2019-07-13 by marcusmueller