gnuradio/gnuradio #2432: VOLK boost::filesystem::path::parent_path() const

Red-Owl, 2019-04-01:
/usr/bin/ld: CMakeFiles/volk_profile.dir/volk_profile.cc.o: in function `write_results(std::vector<volk_test_results_t, std::allocator<volk_test_results_t> > const*, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
volk_profile.cc:(.text+0x31a): undefined reference to `boost::filesystem::path::parent_path() const'
/usr/bin/ld: volk_profile.cc:(.text+0x329): undefined reference to `boost::filesystem::detail::status(boost::filesystem::path const&, boost::system::error_code*)'
/usr/bin/ld: volk_profile.cc:(.text+0x36d): undefined reference to `boost::filesystem::path::parent_path() const'
/usr/bin/ld: volk_profile.cc:(.text+0x4ed): undefined reference to `boost::filesystem::path::parent_path() const'
/usr/bin/ld: volk_profile.cc:(.text+0x4f7): undefined reference to `boost::filesystem::detail::create_directories(boost::filesystem::path const&, boost::system::error_code*)'
collect2: error: ld returned 1 exit status
make[2]: *** [apps/CMakeFiles/volk_profile.dir/build.make:116: apps/volk_profile] Error 1
make[1]: *** [CMakeFiles/Makefile2:327: apps/CMakeFiles/volk_profile.dir/all] Error 2
make: *** [Makefile:141: all] Error 2

not compile volk

michaelld, 2019-04-01:
Can you attach a full build log to this issue or via pastbin.com of the equivalent? Your error snippet here just does't provide enough info! It would also be useful to know the OS & its version you're using, as well as the full CMake configuration commandline and debug output.

Red-Owl, 2019-04-03:
the error is in new version of boost 
Version 1.70.0 beta 1
debian 9 unstable

michaelld, 2019-04-03:
OK; that helps. I know of nobody else running Boost 1.70.0b1 yet ... we usually wait until the release is out to fix issues, since things sometimes change between the betas & release. All of that said, a full build log would still be useful ... or at least snippets showing the commands being issued; we really can't do much without -something- more!

noc0lour, 2019-04-03:
Hm, weird. `parent_path()` is still available in Boost 1.70 and as a matter of fact is not deprecated. We recently switched from branch_path to parent_path for that reason.

Red-Owl, 2019-04-03:
if you tell me which commands to give I give you the outputs

mayorbobster, 2019-04-18:
Hello,
This problem also occurs with boost 1.70 (April 12th, 2019)
I reverted back to 1.69 and gnuradio builds ok (it was a bit messy because when I installed a new version of boost I also installed a new version of cmake).  I kept the new version of cmake when I went back to 1.69, and its cmake-3.14.2 .   
O/S:Linux freedom 5.1.0-rc5 #1 SMP Sun Apr 14 19:41:52 MDT 2019 x86_64 x86_64 x86_64 GNU/Linux
There is a problem with ftp in this kernel, but that's an oops for kernel.org.
I won't post my full build script, or all of the output, but the cmake line I use is:
cmake   -DPYTHON_EXECUTABLE=/usr/bin/python -DQT_LIBRARIES=/usr/lib/libqwt.so ../
and /usr/bin/python -V gives: Python 2.7.15rc1
<forgive the long output here, from 1.70 >
[ 18%] Shebangin gr_plot_fft
[ 18%] Generating __init__.pyc, cmakefile_editor.pyc, code_generator.pyc, grc_yaml_generator.pyc, parser_cc_block.pyc, scm.pyc, util_functions.pyc
/data/gnuradio/gnuradio.git/build/volk/lib/volk_16ic_magnitude_16i_a_orc_impl.c: In function ‘_backup_volk_16ic_magnitude_16i_a_orc_impl’:
/data/gnuradio/gnuradio.git/build/volk/lib/volk_16ic_magnitude_16i_a_orc_impl.c:374:16: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
        if (tmp == 0x80000000 && !(var52.i&0x80000000)) tmp = 0x7fffffff;
                ^~
[ 18%] Built target pygen_gr_utils_python_utils_d13e2

...

[ 21%] Built target wavelet_swig_swig_doc
[ 21%] Shebangin g723_40_audio_loopback.py
/data/gnuradio/gnuradio.git/build/volk/lib/volk_32fc_s32f_magnitude_16i_a_orc_impl.c: In function ‘_backup_volk_32fc_s32f_magnitude_16i_a_orc_impl’:
/data/gnuradio/gnuradio.git/build/volk/lib/volk_32fc_s32f_magnitude_16i_a_orc_impl.c:352:16: warning: comparison between signed and unsigned integer expressions [-Wsign-compare]
        if (tmp == 0x80000000 && !(var50.i&0x80000000)) tmp = 0x7fffffff;
                ^~
[ 21%] Built target pygen_gr_vocoder_examples_9f74c
Scanning dependencies of target pygen_gr_wavelet_python_wavelet_8539a
[ 21%] Building C object volk/lib/CMakeFiles/volk_obj.dir/volk_32fc_s32fc_multiply_32fc_a_orc_impl.c.o

...
[ 25%] Linking CXX executable volk_profile
CMakeFiles/volk_profile.dir/volk_profile.cc.o: In function `write_results(std::vector<volk_test_results_t, std::allocator<volk_test_results_t> > const*, bool, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
volk_profile.cc:(.text+0x286): undefined reference to `boost::filesystem::path::parent_path() const'
volk_profile.cc:(.text+0x295): undefined reference to `boost::filesystem::detail::status(boost::filesystem::path const&, boost::system::error_code*)'
volk_profile.cc:(.text+0x2db): undefined reference to `boost::filesystem::path::parent_path() const'
volk_profile.cc:(.text+0x45b): undefined reference to `boost::filesystem::path::parent_path() const'
volk_profile.cc:(.text+0x465): undefined reference to `boost::filesystem::detail::create_directories(boost::filesystem::path const&, boost::system::error_code*)'
collect2: error: ld returned 1 exit status
volk/apps/CMakeFiles/volk_profile.dir/build.make:115: recipe for target 'volk/apps/volk_profile' failed
make[2]: *** [volk/apps/volk_profile] Error 1
CMakeFiles/Makefile2:318: recipe for target 'volk/apps/CMakeFiles/volk_profile.dir/all' failed
make[1]: *** [volk/apps/CMakeFiles/volk_profile.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 25%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/high_res_timer.cc.o
[ 25%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/io_signature.cc.o
[ 25%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/local_sighandler.cc.o
/data/gnuradio/gnuradio.git/gnuradio-runtime/lib/local_sighandler.cc: In destructor ‘gr::local_sighandler::~local_sighandler()’:
/data/gnuradio/gnuradio.git/gnuradio-runtime/lib/local_sighandler.cc:58:43: warning: throw will always call terminate() [-Wterminate]
       throw std::runtime_error("sigaction");
                                           ^
/data/gnuradio/gnuradio.git/gnuradio-runtime/lib/local_sighandler.cc:58:43: note: in C++11 destructors default to noexcept
[ 25%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/logger.cc.o
[ 25%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/message.cc.o

...

[ 29%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/controlport/rpcserver_booter_aggregator.cc.o
[ 29%] Building CXX object gnuradio-runtime/lib/CMakeFiles/gnuradio-runtime.dir/controlport/rpcserver_selector.cc.o
In file included from /usr/local/include/boost/random/detail/integer_log2.hpp:19:0,
                 from /usr/local/include/boost/random/detail/large_arithmetic.hpp:19,
                 from /usr/local/include/boost/random/detail/const_mod.hpp:23,
                 from /usr/local/include/boost/random/linear_congruential.hpp:30,
                 from /usr/local/include/boost/random/additive_combine.hpp:27,
                 from /usr/local/include/boost/random.hpp:36,
                 from /data/gnuradio/gnuradio.git/gnuradio-runtime/lib/../include/gnuradio/random.h:30,
                 from /data/gnuradio/gnuradio.git/gnuradio-runtime/lib/math/random.cc:42:
/usr/local/include/boost/pending/integer_log2.hpp:7:59: note: #pragma message: This header is deprecated. Use <boost/integer/integer_log2.hpp> instead.
[... cut ...]
...............................................
Reverting to 1.69,  I get a log of messages about:
BOOST_HEADER_DEPRECATED("<boost/integer/common_factor.hpp>");
BOOST_HEADER_DEPRECATED("<boost/integer/common_factor_rt.hpp>");
BOOST_HEADER_DEPRECATED("<boost/integer/common_factor_ct.hpp>");
but it ends with:
[100%] Generating trellis_swig1.pyo
[100%] Generating trellis_swig1.pyc
[100%] Built target pygen_gr_trellis_swig_f6294

Build Successful

...

Install the project...
-- Install configuration: "Release"

jdemel, 2019-04-26:
This issue should be moved to the VOLK issue tracker. Boost is only used in one single file in VOLK, namely 'volk_profile'. And this is even a stand-alone app for VOLK. Nothing should link against it. Thus, 'libvolk' itself is not linked to Boost and GR should only be linked to 'libvolk'.

@mayorbobster could you try to compile VOLK without GNU Radio? I could only see one error in your output. But it's hard to tell. The error I found is:

> [ 25%] Linking CXX executable volk_profile
> CMakeFiles/volk_profile.dir/volk_profile.cc.o: In function `write_results(std::vector<volk_test_results_t, std::allocator<volk_test_results_t> > const*, [...]

The actual errors seems to be:
1. undefined reference to `boost::filesystem::path::parent_path() const'
2. undefined reference to `boost::filesystem::detail::create_directories(boost::filesystem::path const&, boost::system::error_code*)'

This does hint at the fact that the linker does not find Boost filesystem.

A vastly different approach to fix this issue would be to move away from Boost entirely and use STL filesystem. Since this is an issue with a very recent Boost version, I'd expect people who use such a Boost version also use the most recent compilers. In such a case, we could use C++17 with `filesystem`. 

Depending on GCC/Clang versions you'd do `#include <filesystem>` or `#include <experimental/filesystem>` instead of those Boost includes. Next you'd change the namespace definition. That should be it. Now Boost should not be required anymore. In some cases the compiler needs to be told to explicitly link against `filesystem`.

michaelld, 2019-04-26:
Yes this is a Volk issue, not a GR one. I'm going to close this & open a new Volk issue & note this issue in it.

closed 2019-04-26 by michaelld