gnuradio/gnuradio #2436: [feature_request]GRC:  remote management of gnuradio runtime

derekjchen, 2019-04-04:
Hi everyone,

This is a discussion of a possible feature of separating runtime and the UI.

Not sure if github issue tracker is better for this or gnuradio's mailing list.
If I'm in the wrong place, feel free to ask me to post this somewhere else.

Here's my situation:
1. I have a quite a few devices running GR, on different frequencies, and connecting with other part of a bigger system.
2. I need to either NOMACHINE into these machines, or have a monitor connected to these machines when something goes wrong with GR.
3. the number of devices I'm managing will continue to grow.

I'm thinking about if this is the right approach. Branch off gnuradio, and start separating GRC with GR runtime. so that when I launch GRC on my laptop, different GR runtime instances on different machines can be connected to from my GRC.
goal:
1. user still design flowgraph in GRC, and user can pick and choose which machine to run the graph on
2. user can see a list of machine nodes in a new GRC window, and choose to see the details of execution of the flowgraph
3. user can remote monitor machine resources during execution.
4. user can view internode relationship. eg, which node is receiving and/or transmitting at which frequency.

Feedback and thoughts are very welcome.

regards,
Derek

noc0lour, 2019-04-04:
Thanks for reaching out.  Generally the mailinglist `discuss-gnuradio@gnu.org` is better suited to discuss such ideas.
Your idea hits a pain point I think a lot of users with more than one machine using GNU Radio in a "cluster" have. Though the title is a bit misleading. Even right now GRC & gnuradio-runtime are seperated. What is missing is a reasonable RPC mechanism & concept to have multi-machine setups with a semi-central management. 

Looking at the architecture of the current gnuradio-runtime it would be *a lot* of work to implant such a concept to the sample-processing core.

With current GNU Radio I'd personally implement your application separately *using* GNU Radio on all of the nodes "just" as execution environment and implement a control-plane yourself using e.g. Python with some RPC library. If you are motivated you can even extend GRC to support all the features you need with your custom application.

derekjchen, 2019-04-04:
thanks Andrej for replying.
I think implementing a control-plane first to try things out is a great idea!
oh it's separated already? so that means it's possible to run a flowgraph without launching GRC? or you meant runtime has no deps on GRC?

regards,
Derek

noc0lour, 2019-04-04:
@derekjchen if you build a flowgraph in GRC you are basically generating a `xml/yaml` description of your graph. GRC then generates a python (or C++) application with necessary calls to setup that graph programmatically.
You can either use the GRC generated flowgraph as base and go crazy and develop your own application. Or you start from scratch and build your flowgraph entirely using Python. 
https://wiki.gnuradio.org/index.php/Guided_Tutorial_GNU_Radio_in_Python#3.1.1._GRC_Generated_Python_Files

derekjchen, 2019-04-04:
awesome! got it.

closed 2019-04-04 by derekjchen