gnuradio/gnuradio #2439: GRC: QSS Theme not working in 3.8

drmpeg, 2019-04-05:
Here's the error:
```
  File "/home/re/xfer/dvbs2_tx.py", line 252
    tb.setStyleSheetFromFile(/opt/gnuradio-3.7.12git/share/gnuradio/themes/dark.qss)
                             ^
SyntaxError: invalid syntax
```

noc0lour, 2019-04-05:
looks like it's not a string anymore?

drmpeg, 2019-04-05:
I used the file finder dialog. BTW, when the file finder dialog opens, you get a large PyGTK warning.

```
/opt/gnuradio-3.7.12git/lib/python3.5/dist-packages/gnuradio/grc/gui/ParamWidgets.py:327: PyGTKDeprecationWarning: Using positional arguments with the GObject constructor has been deprecated. Please specify keyword(s) for "title, parent, action, buttons" or use a class specific constructor. See: https://wiki.gnome.org/PyGObject/InitializerDeprecations
  transient_for=self._transient_for,
/opt/gnuradio-3.7.12git/lib/python3.5/dist-packages/gnuradio/grc/gui/ParamWidgets.py:327: PyGTKDeprecationWarning: The "buttons" argument must be a Gtk.ButtonsType enum value. Please use the "add_buttons" method for adding buttons. See: https://wiki.gnome.org/PyGObject/InitializerDeprecations
  transient_for=self._transient_for,
/usr/lib/python3/dist-packages/gi/overrides/Gtk.py:537: PyGTKDeprecationWarning: The keyword(s) "parent" have been deprecated in favor of "transient_for" respectively. See: https://wiki.gnome.org/PyGObject/InitializerDeprecations
  self._init(*args, **new_kwargs)
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
```

dl1ksv, 2019-04-13:
Possible fix for error and depreciation warnings in #2447

drmpeg, 2019-04-13:
Thanx for fixing the PyGTK warning. Very nice for those of us that start gnuradio-companion on the command line.

If you're interested in fixing another one, try right clicking in the console window of GRC. Here's the warning I get.
```
/opt/gnuradio-3.7.12git/lib/python3.5/dist-packages/gnuradio/grc/gui/Dialogs.py:144: PyGTKDeprecationWarning: Using positional arguments with the GObject constructor has been deprecated. Please specify keyword(s) for "label" or use a class specific constructor. See: https://wiki.gnome.org/PyGObject/InitializerDeprecations
  lock = Gtk.CheckMenuItem("Scroll Lock")
/opt/gnuradio-3.7.12git/lib/python3.5/dist-packages/gnuradio/grc/gui/Dialogs.py:149: PyGTKDeprecationWarning: Using positional arguments with the GObject constructor has been deprecated. Please specify keyword(s) for "label" or use a class specific constructor. See: https://wiki.gnome.org/PyGObject/InitializerDeprecations
  save = Gtk.ImageMenuItem(Gtk.STOCK_SAVE)
/opt/gnuradio-3.7.12git/lib/python3.5/dist-packages/gnuradio/grc/gui/Dialogs.py:153: PyGTKDeprecationWarning: Using positional arguments with the GObject constructor has been deprecated. Please specify keyword(s) for "label" or use a class specific constructor. See: https://wiki.gnome.org/PyGObject/InitializerDeprecations
  clear = Gtk.ImageMenuItem(Gtk.STOCK_CLEAR)
```

dl1ksv, 2019-04-14:
@drmpeg done. See additional commit in #2447

drmpeg, 2019-04-19:
Fixed with commit https://github.com/gnuradio/gnuradio/commit/8bcb1973acc148c8975ea5386ad52a5e79f327ba

closed 2019-04-19 by drmpeg