gnuradio/gnuradio #2440: Selector bug when a python block is added

Ibr7mm, 2019-04-07:
In the current version of GNU radio, the flow graph stops working completely when using a selector with a python block. At first, this bug thought to be because of an error in the python code. However,  when it was tested with the default python block connected to nulls, it gave the same error. When the python block path is disabled, the selector works perfectly when changing the index between ‘1’ and ‘0’. However, when the python block path is enabled, the system stops working when changing the index to ‘1’ or ‘0’.

[image]

dkozel, 2019-04-07:
Hi @Ibr7mm, What version are you running? Have you modified the Python code in the Embedded Python Block at all or is it still the default template?

Ibr7mm, 2019-04-07:
Hi @dkozel
I'm using GNU Radio Companion 3.7.11
The Embedded Python Block is the default one
The .grc file:
https://drive.google.com/file/d/1TBYloOStpslvLRgWa3CCGBF2mEJ-7H1Y/view?usp=sharing

marcusmueller, 2019-04-21:
The selector block is deprecated, so we won't be fixing it. However, any good replacement would be welcome! So, I'm opening a new issue: Implementing a replacement #2450

closed 2019-04-21 by marcusmueller