gnuradio/gnuradio #2442: gr-uhd AGC support

dwd-pete, 2019-04-08:
UHD currently has the ability to turn AGC on/off via this function: https://files.ettus.com/manual/classuhd_1_1usrp_1_1multi__usrp.html#abdab1f6c3775a9071b15c9805f866486

I am hoping that this functionality gets exposed in gr-uhd, so that I can toggle AGC on/off using GNU Radio.

777arc, 2019-04-09:
Looks like someone may have started that effort - https://lists.gnu.org/archive/html/discuss-gnuradio/2017-10/msg00112.html

Or perhaps there is an underlying reason why we don't want UHD's AGC easily available, like lack of robustness across a range of applications (AGC settings are usually very application specific).

dkozel, 2019-04-09:
Exposing the AGC in gr-uhd would be a fine addition. @dwd-pete I can review that addition if you're able to write it and put up a PR. The changes would be virtually identical, and even slightly simpler as there's no new datatypes or classes involved, as the filter API changes in my (overdue) PR.
https://github.com/gnuradio/gnuradio/pull/2200/files

dwd-pete, 2019-04-12:
I took a stab at the core functionality and it wasn’t too difficult. That said, I don't have a plethora of USRPs to test against, but the underlying code changes are somewhat easy to follow and (as long as set_rx_agc() does what I think it does) should work just fine. I hope to have this functionality pushed in the next few days at the latest.

What I am struggling with is how to update the block in GNU Radio Companion, or if this is something we want to do at all.

My concerns stem from the current default behavior of how the Python code gets generated from the source block. Namely, currently there is no call to set_rx_agc() in the generated code, so (per the UHD docs) the hardware starts up in a "default" configuration, which I'm assuming is off for most devices (otherwise any calls to set the gain would be ignored).

My initial plan was to have an additional drop down menu in the source block (under the "RF Options" tab) entitled "AGC Mode" with two options: "Enabled" and "Disabled". The default would be "Disabled", but then I'm stuck on how to use this information to generate code. For "Enabled" it's obvious I should be calling set_rx_agc() and passing in True, but for "Disabled" if I call set_rx_agc() and pass in False, I will be changing how the block currently generates code (i.e., the block currently assumes AGC is off and generates the calls to set the gain manually). As a result, I'm wondering if I should follow suit and only call set_rx_agc() if it is explicitly being enabled.

Alternatively, we could treat the AGC setting as an "advanced" feature and not include the plumbing to GNU Radio Companion. That is, the user would need to know that they want to enable AGC and have to edit any generated Python code themselves (or write their own Python from scratch). There seems to be a precedent for this, since something like set_recv_timeout() is exposed in the block's source code, but not in GNU Radio Companion (as far as I can tell).

Let me know your thoughts.

dkozel, 2019-04-13:
@dwd-pete Great that you've made a pass at adding the AGC support. 

To conditionally set the agc mode you can use a pattern like this which will only add function call if a value has been chosen.
https://github.com/gnuradio/gnuradio/blob/master/gr-uhd/grc/gen_uhd_usrp_blocks.py#L162-L164

You can use `option_labels` and `options` to provide an "Enabled" option which has a value of "true" so the Python call is straightforward.

Finally, add a callback so any change to the agc mode option will trigger a call to set_rx_agc
https://github.com/gnuradio/gnuradio/blob/master/gr-uhd/grc/gen_uhd_usrp_blocks.py#L201

Finally, all these should be wrapped so that only the Source block has the UI and code for the AGC.
https://github.com/gnuradio/gnuradio/blob/master/gr-uhd/grc/gen_uhd_usrp_blocks.py#L124-L130

dwd-pete, 2019-04-13:
@dkozel I think there's some confusion regarding my question. I wasn't asking how to add the drop down or to use the callbacks, I was asking whether or not set_rx_agc() should be called if "Disabled" is selected. The reason I'm asking is because currently the block assumes manual gain control and only generates the code for the calls to set_rx_gain(). Adding a call to disable AGC prior to these calls would change the current default behavior of the block.  At any rate, thanks for the links.

Also, I am currently stuck on a rather frustrating issue. I am testing with a default install of UHD (installed on my Ubuntu 16.04 machine using "sudo apt-get install libuhd-dev"). This installs version 3.9.2 and this version number is reported when you run uhd_find_devices. My code checks for the UHD_VERSION macro since UHD version 3.9 is the version that added set_rx_agc(), and there is no specific macro for set_rx_agc() (i.e., I can't use something like UHD_USRP_MULTI_USRP_NORMALIZED_GAIN).

However, my check is failing and it seems as if UHD_VERSION is not set. Here's the relevant CMake output (note that the reported UHD version is blank):

> -- Configuring gr-uhd support...
> --   Dependency Boost_FOUND = 1
> --   Dependency UHD_FOUND = TRUE
> --   Dependency ENABLE_GNURADIO_RUNTIME = ON
> --   Dependency ENABLE_GR_FILTER = ON
> --   Dependency ENABLE_GR_BLOCKS = On
> --   Dependency ENABLE_GR_ANALOG = On
> --   Enabling gr-uhd support.
> --   Override with -DENABLE_GR_UHD=ON/OFF
> --   UHD Version: 
> -- Python checking for Mako >= 0.4.2 - found

Any ideas as to what could be causing UHD_VERSION to not be set?

dkozel, 2019-04-13:
Ah. Ideally you wouldn't call it at all. How about adding a "Default" option which doesn't generate any initialization code? Leave in the callback so the value can still be updated at runtime.

Sorry, no answer to the UHD_VERSION off the top of my head. I'd have to checkout the specific version and see exactly what the CMake files were like at that point.

dwd-pete, 2019-04-15:
I got the UHD_VERSION macro to work. See my work in progress here: https://github.com/dwd-pete/gnuradio/commit/74bf3d8929c7642f7c23275f1064ad8461dc7380

Working on adding the hooks to GNU Radio Companion now. I should have some time to work on that this week.

dkozel, 2019-04-16:
Looking good so far. Thanks!

dwd-pete, 2019-05-18:
Closing because pull request was merged.

closed 2019-05-18 by dwd-pete