gnuradio/gnuradio #2443: GRC: tutorial "OutOfTreeModules"

Massi1987, 2019-04-09:
Hi everybody,

I'm new to GNU Radio and currently doing the tutorials of https://wiki.gnuradio.org/index.php/Tutorials

My setup: 
Ubuntu 18.94 LTS running on OracleVirtualBox
GNU Radio Companion installed as described on the wiki site.

I worked with the tutorial "Developing out-of-tree modules and writing blocks" (https://wiki.gnuradio.org/index.php/Tutorials) and did everything step-by-step until I got stuck at point 5.5 "Let's try that -- running make test". When I ran the test, all tests failed. I also used "ctest -V -R square" to get more information about the issue but I can't figure out the reason why it is failing (see terminal output below).

Is something missing in the tutorial which is so obvious that it is not described?
The test in the original tutorial "Guided Tutorial GNU Radio in C++" is also not working (and unfortunately the block is not working in GRC too.)

**Output of the terminal window:**
$ ctest -V -R square
UpdateCTestConfiguration  from :/home/tobias/GR Tutorial/newhoto/gr-howto/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/home/tobias/GR Tutorial/newhoto/gr-howto/build/DartConfiguration.tcl
Test project /home/tobias/GR Tutorial/newhoto/gr-howto/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 2
    Start 2: qa_square_ff

2: Test command: /bin/sh "/home/tobias/GR Tutorial/newhoto/gr-howto/build/python/qa_square_ff_test.sh"
2: Test timeout computed to be: 9.99988e+06
2: /home/tobias/GR Tutorial/newhoto/gr-howto/build/python/qa_square_ff_test.sh: 5: export: Tutorial/newhoto/gr-howto/build/python:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin: bad variable name
1/1 Test #2: qa_square_ff .....................***Failed    0.00 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.00 sec

The following tests FAILED:
	  2 - qa_square_ff (Failed)
Errors while running CTest

marcusmueller, 2019-04-21:
When you look closely, you'll notice that `export` is complaining that "half a path" is not a good variable name. That's because our testing infrastructure writes a shell script that contains a line

```
export PYTHONPATH=/path/to/stuff:/path/to/OOT Module/build/python
```

Now, when you look at that, you'll notice that the space breaks that line, because instead of

```ENV_VARIABLE=value```

`export` now sees 

```ENV_VARIABLE=value MORE_STUFF```

and is confused.

To fix this for *you* is easy: Don't use a path with a space in it.

To generally fix that, maybe adding quotation marks around the `export` argument suffices, maybe not (depends on Python, too). Needs investigation.

mormj, 2019-06-18:
This problem also exists for doxygen strings when the OOT path has a space in it.  

For the qa...sh file that is generated, it looks like this can be fixed by converting strings from 
path/to/src\ dir/
to 
"path/to/src dir/"

in GrTest.cmake such as:
    file(TO_NATIVE_PATH "${CMAKE_CURRENT_SOURCE_DIR}" srcdir)
    string(CONCAT srcdir "\"" ${srcdir} "\"")
    string(REPLACE "\\ " " " srcdir ${srcdir})

Will clean this up, but it seems this could easily break some other things and definitely needs to be tried in Windows.

PR #2638 (2019-07-19): modtool: put proper quotes when unescaped shell paths
Adds quotes to paths to prevent issues when the
user has a space in their OOT shell paths

Slight change to #2560 which should prevent CI from
broadly failing

Fixes #2443
commit 6984088e (2019-07-19): modtool: put proper quotes when unescaped shell paths

Adds quotes to paths to prevent issues when the
user has a space in their OOT shell paths

Slight change to #2560 which should prevent CI from
broadly failing

Fixes #2443
closed 2019-07-19 by marcusmueller