gnuradio/gnuradio #2445: Problems with gnuradio in ubuntu 18.04

alvals, 2019-04-12:
Dear, I hope you help me with this problem.

I need to build gnuradio from the source, I tried to do it manually but it is very complicated for a beginner like me, many mistakes are generated along the way.

Anyway, now I'm trying it with Pybomb and I followed the recipe below (all in a clean installation of Ubuntu 18.04):

`sudo apt-get install python-pip`
`sudo pip install -U pip`
`sudo pip install PyBOMBS`

`sudo pip install PyBOMBS`

Initiating the default prefix:
`sudo pybombs prefix init /usr/local -a usrlocal`
`sudo pybombs config default_prefix /usr/local`

Adding recipes:
`sudo pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git`
`sudo pybombs recipes add gr-etcetera git+https://github.com/gnuradio/gr-etcetera.git`

Installing dependencies and GNURadio:
`sudo pybombs install gnuradio`

...everything installed without incident until the part of the dependencies and began to install GNURadio, then it generated the following error:

$ sudo pybombs install gnuradio
PyBOMBS.ConfigManager - INFO - Prefix Python version is: 2.7.15
PyBOMBS - INFO - PyBOMBS Version 2.3.3
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
PyBOMBS.install_manager - INFO - Phase 1: Creating install tree and installing binary packages:
Install tree:
|
\- gnuradio
   |
   +- uhd
   |
   \- apache-thrift
PyBOMBS.install_manager - INFO - Phase 2: Recursively installing source packages to prefix:
PyBOMBS.install_manager - INFO - Installing package: apache-thrift
PyBOMBS.Packager.apt - INFO - Install python-apt to speed up apt processing.
Building:    (100%) [=================================================================================================================]
make  all-recursive
make[1]: Entering directory '/usr/local/src/apache-thrift'
Making all in lib/cpp
make[2]: Entering directory '/usr/local/src/apache-thrift/lib/cpp'
Making all in .
make[3]: Entering directory '/usr/local/src/apache-thrift/lib/cpp'
make[3]: Nothing to be done for 'all-am'.
make[3]: Leaving directory '/usr/local/src/apache-thrift/lib/cpp'
make[2]: Leaving directory '/usr/local/src/apache-thrift/lib/cpp'
Making all in compiler/cpp
make[2]: Entering directory '/usr/local/src/apache-thrift/compiler/cpp'
Making all in src
make[3]: Entering directory '/usr/local/src/apache-thrift/compiler/cpp/src'
make  all-am
make[4]: Entering directory '/usr/local/src/apache-thrift/compiler/cpp/src'
make[4]: Nothing to be done for 'all-am'.
make[4]: Leaving directory '/usr/local/src/apache-thrift/compiler/cpp/src'
make[3]: Leaving directory '/usr/local/src/apache-thrift/compiler/cpp/src'
Making all in src/thrift/plugin
make[3]: Entering directory '/usr/local/src/apache-thrift/compiler/cpp/src/thrift/plugin'
make  all-am
make[4]: Entering directory '/usr/local/src/apache-thrift/compiler/cpp/src/thrift/plugin'
make[4]: Nothing to be done for 'all-am'.
make[4]: Leaving directory '/usr/local/src/apache-thrift/compiler/cpp/src/thrift/plugin'
make[3]: Leaving directory '/usr/local/src/apache-thrift/compiler/cpp/src/thrift/plugin'
Making all in .
make[3]: Entering directory '/usr/local/src/apache-thrift/compiler/cpp'
make[3]: Nothing to be done for 'all-am'.
make[3]: Leaving directory '/usr/local/src/apache-thrift/compiler/cpp'
make[2]: Leaving directory '/usr/local/src/apache-thrift/compiler/cpp'
Making all in lib
make[2]: Entering directory '/usr/local/src/apache-thrift/lib'
Making all in json
make[3]: Entering directory '/usr/local/src/apache-thrift/lib/json'
make[4]: Entering directory '/usr/local/src/apache-thrift/lib/json'
make[4]: Nothing to be done for 'all-am'.
make[4]: Leaving directory '/usr/local/src/apache-thrift/lib/json'
make[3]: Leaving directory '/usr/local/src/apache-thrift/lib/json'
Making all in xml
make[3]: Entering directory '/usr/local/src/apache-thrift/lib/xml'
make[4]: Entering directory '/usr/local/src/apache-thrift/lib/xml'
make[4]: Nothing to be done for 'all-am'.
make[4]: Leaving directory '/usr/local/src/apache-thrift/lib/xml'
make[3]: Leaving directory '/usr/local/src/apache-thrift/lib/xml'
Making all in c_glib
make[3]: Entering directory '/usr/local/src/apache-thrift/lib/c_glib'
Making all in .
make[4]: Entering directory '/usr/local/src/apache-thrift/lib/c_glib'
make[4]: Nothing to be done for 'all-am'.
make[4]: Leaving directory '/usr/local/src/apache-thrift/lib/c_glib'
Making all in test
make[4]: Entering directory '/usr/local/src/apache-thrift/lib/c_glib/test'
make  all-recursive
make[5]: Entering directory '/usr/local/src/apache-thrift/lib/c_glib/test'
make[6]: Entering directory '/usr/local/src/apache-thrift/lib/c_glib/test'
make[6]: Nothing to be done for 'all-am'.
make[6]: Leaving directory '/usr/local/src/apache-thrift/lib/c_glib/test'
make[5]: Leaving directory '/usr/local/src/apache-thrift/lib/c_glib/test'
make[4]: Leaving directory '/usr/local/src/apache-thrift/lib/c_glib/test'
make[3]: Leaving directory '/usr/local/src/apache-thrift/lib/c_glib'
Making all in py
make[3]: Entering directory '/usr/local/src/apache-thrift/lib/py'
### **/usr/bin/python3 setup.py build**
Error processing line 1 of /usr/lib/python3/dist-packages/zope.interface-4.3.2-nspkg.pth:

Failed to import the site module
Traceback (most recent call last):
  File "/usr/lib/python3.6/site.py", line 174, in addpackage
    exec(line)
  File "<string>", line 1, in <module>
  File "/usr/lib/python3.6/types.py", line 171, in <module>
    import functools as _functools
  File "/usr/lib/python3.6/functools.py", line 21, in <module>
    from collections import namedtuple
  File "/usr/lib/python3.6/collections/__init__.py", line 32, in <module>
    from reprlib import recursive_repr as _recursive_repr
  File "/usr/local/lib/python2.7/dist-packages/reprlib/__init__.py", line 7, in <module>
    raise **ImportError('This package should not be accessible on Python 3. '
ImportError: This package should not be accessible on Python 3. Either you are trying to run from the python-future src folder or your installation of python-future is corrupted.**

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/lib/python3.6/site.py", line 570, in <module>
    main()
  File "/usr/lib/python3.6/site.py", line 557, in main
    known_paths = addsitepackages(known_paths)
  File "/usr/lib/python3.6/site.py", line 345, in addsitepackages
    addsitedir(sitedir, known_paths)
  File "/usr/lib/python3.6/site.py", line 213, in addsitedir
    addpackage(sitedir, name, known_paths)
  File "/usr/lib/python3.6/site.py", line 184, in addpackage
    import traceback
  File "/usr/lib/python3.6/traceback.py", line 3, in <module>
    import collections
  File "/usr/lib/python3.6/collections/__init__.py", line 32, in <module>
    from reprlib import recursive_repr as _recursive_repr
  File "/usr/local/lib/python2.7/dist-packages/reprlib/__init__.py", line 7, in <module>
    raise ImportError('This package should not be accessible on Python 3. '
ImportError: This package should not be accessible on Python 3. Either you are trying to run from the python-future src folder or your installation of python-future is corrupted.
Makefile:626: recipe for target 'py3-build' failed
make[3]: *** [py3-build] Error 1
make[3]: Leaving directory '/usr/local/src/apache-thrift/lib/py'
Makefile:564: recipe for target 'all-recursive' failed
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory '/usr/local/src/apache-thrift/lib'
Makefile:646: recipe for target 'all-recursive' failed
make[1]: *** [all-recursive] Error 1
make[1]: Leaving directory '/usr/local/src/apache-thrift'
Makefile:560: recipe for target 'all' failed
make: *** [all] Error 2
PyBOMBS.Packager.source - ERROR - Build failed. See output above for error messages.
PyBOMBS.Packager.source - ERROR - Problem occurred while building package apache-thrift:
Build failed.
PyBOMBS.install_manager - ERROR - Error installing package apache-thrift. Aborting.

I understand that Ubuntu 18.04 comes by default with Python 3.6.7 and I think that is the problem. I tried to solve it with Pyenv and change the global version of Python to 2.7 but the problem is still present. Also attached some characteristics of the distribution that I have:

- OS: Ubuntu 18.04.2 LTS x86_64 
- Host: G3
- Kernel: 4.15.0-47-generic  
- Packages: 2071
- Shell: bash 4.4.19  
- DE: GNOME 3.28.3 
- WM: GNOME Shell  
- Theme: Ambiance [GTK2/3] 
- Icons: Ubuntu-mono-dark [GTK2/3] 
- Terminal: gnome-terminal 
- CPU: Intel i7-8750H (12) @ 4.100GHz 
- Memory: 2257MiB / 7835MiB 

Additional information that i think it help:

$ which python
/home/alvaroloa/.pyenv/shims/python

$ which python3
/usr/bin/python3

$ which python2.7
/home/alvaroloa/.pyenv/shims/python2.7

$ which python2
/home/alvaroloa/.pyenv/shims/python2

$ python --version
Python 2.7.15rc1

aru31, 2019-04-12:
Hi @alvals, Please go through this [section](https://github.com/gnuradio/gnuradio#pybombs-with-support-of-python-3x) of README.md if you require to build GNU Radio with pyBOMBS along with python3 support.
Hope this helps. :)

Teque5, 2019-04-23:
If you follow the steps, it works in 18.04. You missed step 4:
> In `gnuradio.lwr` recipe file, add "-DENABLE_CTRLPORT_THRIFT=OFF" in "config_opt".

alvals, 2019-05-13:
Hi everyone
I did not want to use Python 3 due to possible compatibility problems (maybe). So I used the home directory with Python 2 and the build process was completed without errors but in the process, I could see many warnings about several modules like OBSOLETE (I do not know if I should worry) besides that QTgui does not work at all well, especially with Frequency sink.

Well, these are the steps I followed, please if someone sees something that I missed, please tell me:

`$ sudo apt-get install python-pip`
`$ sudo pip install PyBOMBS`
`$ pybombs recipes add gr-recipes git+https://github.com/gnuradio/gr-recipes.git`
`$ pybombs recipes add gr-etcetera git+https://github.com/gnuradio/gr-etcetera.git`

`$ pybombs prefix init ~/prefix -a myprefix`
`$ pybombs install -v -v -v gnuradio`

Run GNU Radio Companion from the new prefix:
`$ pybombs run gnuradio-companion`

RUN:
Now, when using an rtl-sdr device, after installing gr-osmosdr, I must execute: `pybombs run gnuradio-companion` with SUDO, otherwise, RTL does not recognize (I do not know why)

Another thing that I observed is that Pybombs install by default the Maint-3.7 branch, should I install the Master branch? and if so, how do I do it? I tried to manually modify the recipe gnuradio.lwr of Pybombs but errors appear in the build process.

Thanks, it seems that compiling in a directory of prefixes is a better option to have several gnuradio versions. With this in mind, I will try to install GNURadio with Python 3 without fear of conflicts with the other version.

Teque5, 2019-05-13:
@alvals I have gnuradio `master` (3.8) installed in Ubuntu 18.04 via pybombs.

I don't have access to my notes at this exact second, but the simplest thing to do is just forget python2. So do `pip3 install`, you have to modify the reciple lwr to point to `master`, and you also have to change your `PYTHONPATH` variable from `2.6` & `2.7` to `python3` & `python3.6`. UHD & GNUradio work, and many OOT modules work, but several need modified cmake files to install correctly. One sticking point is if you need `gr-osmosdr`, then someone needs to fix `gr-iqbal`, since that seems to be broken for `3.8`.

I've got a OOT module conversion checklist I'm working on, but it's not complete yet.

noc0lour, 2019-05-16:
Please move support and discussion to our mailing list <discuss-gnuradio@gnu.org>. This issue tracker is only to track issues with the GNU Radio code itself.

closed 2019-05-16 by noc0lour