gnuradio/gnuradio #2446: Fails to build with boost 1.70: error: no member named 'get_io_service' in 'boost::asio::basic_socket_acceptor<boost::asio::ip::tcp, boost::asio::executor>'

yurivict, 2019-04-12:
See the downstream bug report here: https://bugs.freebsd.org/bugzilla/show_bug.cgi?id=236601

michaelld, 2019-04-19:
Can you test the solution here in PR #2451 ? It hopefully addresses this issue. Thx! NP if not; we're working our way up testing / verification ourselves, but it'll take a bit due to getting Boost 1.70.0 installed and its new CMake scripts working correctly.

ffontaine, 2019-04-29:
Personally, I disabled the installation of the boost cmake scripts through --no-cmake-config. For me, they are just not ready for production. We had a lot build failures on buildroot due to these scripts being used instead of the "standard" cmake ones. For example, most of the boost packages that I know (host-thrift, domoticz, i2pd ...) will fail because BOOST_INCLUDE_DIRS will be empty.

michaelld, 2019-04-29:
@ffontaine I'm coming to that same conclusion: the Boost CMake scripts are just not ready for production yet. I'm going to remove them (just rm -rf post-install), fix the install flag you note & hopefully that will allow for clean configuration & building of GR / UHD / Volk & others that use Boost.

scpeters, 2019-06-02:
I think this is fixed by #2500

closed 2019-06-02 by yurivict

venkateshsathya, 2019-08-15:
Hi Michael,

I am attempting to build GNU radio SW v3.7.13.5 . I followed the instructions below and noticed error message similar to what is posted in this page. I also see a potential solution posted. However, I am unable to figure out how to use this solution. Kindly help me with the issue. 

Instructions used: 

   mkdir workarea-gnuradio
   cd workarea-gnuradio
   git clone --recursive https://github.com/gnuradio/gnuradio
   cd gnuradio
   git checkout v3.7.13.5
   git submodule update --init --recursive
$ mkdir build
$ cd build
$ CC=/usr/bin/llvm-gcc CXX=/usr/bin/llvm-g++ cmake -DPYTHON_EXECUTABLE=/opt/local/bin/python2.7 -DPYTHON_INCLUDE_DIR=/opt/local/Library/Frameworks/Python.framework/Versions/2.7/Headers -DPYTHON_LIBRARY=/opt/local/Library/Frameworks/Python.framework/Versions/2.7/Python -DSPHINX_EXECUTABLE=/opt/local/bin/rst2html-2.7.py -DGR_PYTHON_DIR=/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages ..
$ make

Error noticed: 

`[ 26%] Building CXX object gr-blocks/lib/CMakeFiles/gnuradio-blocks.dir/skiphead_impl.cc.o
[ 26%] Building CXX object gr-blocks/lib/CMakeFiles/gnuradio-blocks.dir/socket_pdu_impl.cc.o
/Users/venkat/Documents/ModulationClassificationOTA/workarea-gnuradio/gnuradio/gr-blocks/lib/socket_pdu_impl.cc:168:82: error: no member named
      'get_io_service' in 'boost::asio::basic_socket_acceptor<boost::asio::ip::tcp, boost::asio::executor>'
      tcp_connection::sptr new_connection = tcp_connection::make(d_acceptor_tcp->get_io_service(), d_rxbuf.size(), d_tcp_no_delay);
                                                                 ~~~~~~~~~~~~~~  ^
1 error generated.
make[2]: *** [gr-blocks/lib/CMakeFiles/gnuradio-blocks.dir/socket_pdu_impl.cc.o] Error 1
make[1]: *** [gr-blocks/lib/CMakeFiles/gnuradio-blocks.dir/all] Error 2`