gnuradio/gnuradio #2449: FIR Filter randomly drops samples

JRock007, 2019-04-15:
## Issue
When using FIR filters, some samples are dropped during the filtering. I noticed this using RRC Filter taps (see how to reproduce below).
This causes what looks like a gradual phase shift over time in the output:
[image]
I have not been able to notice a pattern in the drops (e.g. every N samples).

## Environment
Ubuntu 18.04.2 LTS
GNU Radio Companion 3.7.11

## How to reproduce
I built a simple flow graph with the following elements:

- RRC Filter Taps with 16 samples per symbole, and 176 (11*sps) taps. The symbol rate is thus set to samp_rate / sps, and the gain to sps**0.5 to compensate for the filters,
- Interpolating FIR Filter and Decimating FIR Filter using the defined taps and respectively set with interpolation to sps and decimation to sps,
- A (custom made) resampler, which in this case is simply used to drop the first 11 (nbr_taps / sps) samples, as they are not representative since the filters are not "initialised" yet. The issue occurs without this block, but it's simpler to visualise like this,
- A signal source outputting a square signal to see when samples are being dropped,
- And a time sink to compare the source and the filtered output.

[image]

With this setup, the output looks like this:
[image]
At first, the source and output are perfectly synchronised, but after some time they become out of sink. Here's a closeup of the moment this occurs:
[image]
Since it may not be obvious that a sample is being dropped, I used the same flowgraph but simply tweaked the source and time sink parameters to visualise the following:
[image]

I have tried to lower the samp_rate (though it's only at 32k), but it hasn't fixed the issue.
Moreover, I don't think it's an underflow issue since mu CPU is far from being overused.

marcusmueller, 2019-04-21:
Thanks for the bug report. I don't mean to be mean, but typically such bugs really appear in custom written blocks – can you replace your custom block with "skip head", which does exactly what you describe?

JRock007, 2019-04-22:
Thanks for your reply.

After testing with the "skip head" bloc, I don't seem to be able to reproduce the issue anymore. I thought I had replicated the issue without my custom block, but I guessed I hadn't tested properly.

Thanks again for your help!

closed 2019-04-22 by JRock007

marcusmueller, 2019-04-22:
I didn't do anything! If you happen to stumble across this again, please do not hesitate to reopen this issue, and/or to ping us on the mailing list! (if this really happens, it's a severe bug)