gnuradio/gnuradio #2450: CMakeLists for gnuradio-runtime needs Boost::thread

qnarferao, 2019-04-18:
Compilation on macOS 10.14 results in:

> Undefined symbols for architecture x86_64:
>   "boost::this_thread::interruption_point()", referenced from:
>       boost::condition_variable::wait(boost::unique_lock<boost::mutex>&) in pmt_pool.cc.o
>   "boost::detail::get_current_thread_data()", referenced from:
>       boost::detail::interruption_checker::interruption_checker(_opaque_pthread_mutex_t*, _opaque_pthread_cond_t*) in pmt_pool.cc.o
> ld: symbol(s) not found for architecture x86_64

Correct gnuradio-runtime/lib/pmt/CMakeLists.txt:
```
diff --git a/gnuradio-runtime/lib/pmt/CMakeLists.txt b/gnuradio-runtime/lib/pmt/CMakeLists.txt
index c46227e76..9fb98d0fc 100644
--- a/gnuradio-runtime/lib/pmt/CMakeLists.txt
+++ b/gnuradio-runtime/lib/pmt/CMakeLists.txt
@@ -27,6 +27,7 @@ add_library(gnuradio-pmt
 
 target_link_libraries(gnuradio-pmt
   Boost::boost
+  Boost::thread
   Log4Cpp::log4cpp
 )
```

michaelld, 2019-04-19:
Could be; OSX DYLD works differently than Linux's LD so we might need to be explicit in setting that library for linking PMT. We'll verify & get back.

michaelld, 2019-04-19:
I'm currently fighting with Boost 1.70.0's new CMake scripts, which somehow provide API (header) access but fail for ABI (library) access. So, please be patient as I/we work though these various issues.

michaelld, 2019-04-20:
Verified & I've put in a pull request with the fixes. Thanks!

marcusmueller, 2019-04-21:
closed via #2456

closed 2019-04-21 by marcusmueller