gnuradio/gnuradio #2452: Boost 1.7x compat

marcusmueller, 2019-04-18:
@michaelld raised the point that 1.7 doesn't work with master right now (as indicated by bug reports). He'll try locally and report.

michaelld, 2019-04-30:
The good news is that UHD and Volk GIT master both work with Boost 1.70.0 without further modifications, at least in my testing on OSX 10.14 and 10.6. I haven't tested the most recent releases yet, but I'm working my way there. More as I know it.

michaelld, 2019-05-02:
Ref GR issue #2451, which I just closed since it's not the correct solution here. I'll get a PR in place tomorrow with a more robust solution, once I've had time to test / verify on a few different Boost versions.

michaelld, 2019-05-02:
Also: with this fix, GR builds fully using Boost 1.70.0, so this should be the only fix required for building. It is entirely possible that GR (and, possibly, UHD and/or Volk) are using deprecated interfaces that may be removed in some future Boost. We can only future-proof so much ... but we'll do what we can when we find it, as I'll do with the more robust solution I came up with for the boost::asio issue.

michaelld, 2019-05-02:
With my changes, GR builds cleanly using 1.70.0 and also passes make test for all tests that usually pass (e.g., sometimes the zmq tests fail, sometimes they don't). In particular, the socket_pdu passes, and that's where these changes would impact GR.

michaelld, 2019-05-02:
I still need to do testing with Boost back to 1.65.0 ... luckily MacPorts can easily handle those multiple Boost version installs ... just takes a little work & time on my part to get them in place.

PR #2500 (2019-05-28): blocks: simple fix for Boost 1.70.0 in socket_pdu
NOTE: There have been multiple fixes proposed, such as https://github.com/gnuradio/gnuradio/pull/2451 . This one is the simplest and most compatible; it comes from < https://svnweb.freebsd.org/ports/head/comms/gnuradio/files/patch-gr-blocks_lib_socket__pdu__impl.cc?view=markup&pathrev=499093 >. With this fix, GR GIT master builds with Boost 1.70.0.

Fixes #2452
closed 2019-05-28 by mbr0wn