gnuradio/gnuradio #2453: gr-audio for osx needs updated include for audio_registry.h

qnarferao, 2019-04-18:
gr-audio/lib/osx/*.cc that reference audio_registry.h need to reference ../audio_registry.h
Patch:
```
diff --git a/gr-audio/lib/osx/osx_impl.cc b/gr-audio/lib/osx/osx_impl.cc
index f3ed94c65..2a7337511 100644
--- a/gr-audio/lib/osx/osx_impl.cc
+++ b/gr-audio/lib/osx/osx_impl.cc
@@ -24,7 +24,7 @@
 #include "config.h"
 #endif
 
-#include "audio_registry.h"
+#include "../audio_registry.h"
 
 #include <gnuradio/audio/osx_impl.h>
 
diff --git a/gr-audio/lib/osx/osx_sink.cc b/gr-audio/lib/osx/osx_sink.cc
index 8682f834b..80dc5fe5f 100644
--- a/gr-audio/lib/osx/osx_sink.cc
+++ b/gr-audio/lib/osx/osx_sink.cc
@@ -24,7 +24,7 @@
 #include "config.h"
 #endif
 
-#include "audio_registry.h"
+#include "../audio_registry.h"
 #include "osx_sink.h"
 
 #include <gnuradio/io_signature.h>
diff --git a/gr-audio/lib/osx/osx_source.cc b/gr-audio/lib/osx/osx_source.cc
index 0485c5478..48001e263 100644
--- a/gr-audio/lib/osx/osx_source.cc
+++ b/gr-audio/lib/osx/osx_source.cc
@@ -24,7 +24,7 @@
 #include "config.h"
 #endif
 
-#include "audio_registry.h"
+#include "../audio_registry.h"
 #include "osx_source.h"
 
 #include <gnuradio/io_signature.h>

```

michaelld, 2019-04-18:
what version of GR are you providing this patch for?
Does GR not build without your patch?
what are the versions of OSX and Xcode?
how is GR installed?

qnarferao, 2019-04-19:
what version of GR are you providing this patch for?
HEAD

Does GR not build without your patch?
what are the versions of OSX and Xcode?
how is GR installed?

Rather than answer these I'll point you to https://github.com/gnuradio/gnuradio/tree/master/gr-audio/lib where you can see audio_registry.h exists. In any of the other sub-directories (alsa, oss) they correctly point back a directory to this header. The osx/ files never got updated. This patches that.

michaelld, 2019-04-19:
After reviewing the latest changes to this component, I see what you mean. I haven't updated to current GIT master head in a couple months. Let me do that & I'd guess I'll see where your patch is required. Thx!

michaelld, 2019-04-20:
Verified & I've put in a pull request with the fixes. Thanks!

marcusmueller, 2019-04-21:
closed via #2457

closed 2019-04-21 by marcusmueller