gnuradio/gnuradio #2462: gr-utils: Restructure

marcusmueller, 2019-04-21:
gr-utils has the following tree structure:

<pre><font color="#0087FF">gr-utils</font>
├── <font color="#0087FF">octave</font>
└── <font color="#0087FF">python</font>
    ├── <font color="#0087FF">modtool</font>
    └── <font color="#0087FF">utils</font>
</pre>

which makes little to no sense, since nobody cares about which language tools are written in. On the contrary, it makes it harder for us to selectively en- or disable features like modtool based on dependency availability.

Also, `gr-utils/python/utils` is not only a really bad directory name, but should also check whether the required python GUI bindings are available at all.

I'd propose:

<pre><font color="#0087FF">gr-utils</font>
├── <font color="#0087FF">octave</font>
└── <font color="#0087FF">modtool</font>
└── <font color="#0087FF">plot_tools</font>
</pre>

With `filter_designer` being another candidate to be moved here – not really being a part of GRC, where it currently resides.

michaelld, 2019-04-23:
:+1: from me for all of these, including moving the `filter_designer` tool to `gr-utils`.

mackncheesiest, 2019-05-03:
A bit unfamiliar with the codebase, but should migration of the `filter_designer` involve moving both `gr-filter/python/filter/design` and `gr-filter/python/filter/gui` to, say, `gr-utils/filter_design`?

amCap1712, 2020-01-15:
@mackncheesiest Is this open for work?

mackncheesiest, 2020-01-15:
Yeah sure, I don't have the availability at the moment to guide the previous WIP PR through

japm48, 2020-03-26:
I am volunteering to take care of this restructuring if neither @amCap1712 nor @mackncheesiest are available for it.

Is that Ok?
/cc @michaelld

85ayush, 2021-02-23:
I would love to work on this issue, I am new to open source contribution, can you please guide me?

mackncheesiest, 2021-02-23:
I believe this issue was mostly taken care of via https://github.com/gnuradio/gnuradio/pull/3283. I might be wrong, but other than the potential move of `filter_designer` to this section of the tree, this issue can seemingly be closed

marcusmueller, 2021-03-24:
Agreeing to the fullest, @mackncheesiest @85ayush.

closed 2021-03-24 by marcusmueller