gnuradio/gnuradio #2464: import digital_swig error

V18006, 2019-04-22:
Sir,
I am new to GNU radio, so sorry if this error is very basic !!  
I am getting an error while trying one of your codes given in 
[https://github.com/gnuradio/gnuradio/blob/master/gr-digital/python/digital/ofdm_txrx.py](url)
The error was: 
Traceback (most recent call last):
    File "ofdm_txrx.py", line 33, in <module>
      import digital_swig as digital
ImportError: No module named digital_swig
Sir, I used  "python ofdm_txrx.py" to run the code from the terminal.
Should I use any other way to run the code?

marcusmueller, 2019-04-22:
We're not just one sir, there's quite a few ladies here, too! Anyway, what you're describing sounds like an installation problem, not a GNU Radio bug, so I'm closing this issue.

V18006, 2019-04-22:
Sir / Madam,
But all the other blocks are working. What should I do?

michaelld, 2019-04-22:
OK so -some- of GNU Radio is installed properly. When you installed GNU Radio, did you make a note of which components were enabled / disabled? If not, go back and revisit the CMake output, maybe gr-digital was not enabled for some reason?

michaelld, 2019-04-27:
No response for 5 days. I'm closing this issue. If the issue is still relevant for the original poster, please reopen and post more information here to help us diagnose the issue.

closed 2019-04-27 by michaelld