gnuradio/gnuradio #2470: some errors in gr-digital

dl1ksv, 2019-05-07:
Loading example_corr_est.grc from the gr-digital examples in grc leads to a lot of errors.

ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate variable block hdr_const
Traceback (most recent call last):
  File "/usr/local/gnuradio/lib/python3.7/dist-packages/gnuradio/grc/core/FlowGraph.py", line 226, in renew_namespace
    value = eval(variable_block.value, namespace, variable_block.namespace)
  File "<string>", line 1, in <module>
  File "/usr/local/gnuradio/lib/python3.7/dist-packages/gnuradio/digital/digital_swig2.py", line 524, in make
    return _digital_swig2.constellation_calcdist_make(constell, pre_diff_code, rotational_symmetry, dimensionality, normalize_points)
TypeError: in method 'constellation_calcdist_make', argument 1 of type 'std::vector< gr_complex,std::allocator< gr_complex > >'
ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate variable block rrc_taps
Traceback (most recent call last):
  File "/usr/local/gnuradio/lib/python3.7/dist-packages/gnuradio/grc/core/FlowGraph.py", line 226, in renew_namespace
    value = eval(variable_block.value, namespace, variable_block.namespace)
  File "<string>", line 1, in <module>
  File "/usr/local/gnuradio/lib/python3.7/dist-packages/gnuradio/filter/filter_swig.py", line 253, in root_raised_cosine
    return _filter_swig.firdes_root_raised_cosine(gain, sampling_freq, symbol_rate, alpha, ntaps)
TypeError: in method 'firdes_root_raised_cosine', argument 1 of type 'double'
ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate variable block rx_psf_taps1
Traceback (most recent call last):
  File "/usr/local/gnuradio/lib/python3.7/dist-packages/gnuradio/grc/core/FlowGraph.py", line 226, in renew_namespace
    value = eval(variable_block.value, namespace, variable_block.namespace)
  File "<string>", line 1, in <module>
  File "/usr/local/gnuradio/lib/python3.7/dist-packages/gnuradio/filter/filter_swig.py", line 253, in root_raised_cosine
    return _filter_swig.firdes_root_raised_cosine(gain, sampling_freq, symbol_rate, alpha, ntaps)
TypeError: in method 'firdes_root_raised_cosine', argument 2 of type 'double'
ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate variable block modulated_sync_word
Traceback (most recent call last):
  File "/usr/local/gnuradio/lib/python3.7/dist-packages/gnuradio/grc/core/FlowGraph.py", line 226, in renew_namespace
    value = eval(variable_block.value, namespace, variable_block.namespace)
  File "<string>", line 1, in <module>
AttributeError: 'NoneType' object has no attribute 'to_basic_block'
ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate variable block modulated_sync_word
Traceback (most recent call last):
  File "/usr/local/gnuradio/lib/python3.7/dist-packages/gnuradio/grc/core/FlowGraph.py", line 226, in renew_namespace
    value = eval(variable_block.value, namespace, variable_block.namespace)
  File "<string>", line 1, in <module>
AttributeError: 'NoneType' object has no attribute 'to_basic_block'
>>> Done
ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate variable block modulated_sync_word
Traceback (most recent call last):
  File "/usr/local/gnuradio/lib/python3.7/dist-packages/gnuradio/grc/core/FlowGraph.py", line 226, in renew_namespace
    value = eval(variable_block.value, namespace, variable_block.namespace)
  File "<string>", line 1, in <module>
AttributeError: 'NoneType' object has no attribute 'to_basic_block'

Despite those errors grc generates python code, but this code fails to run with:

  File "/home/schroer/gnuradiocomponents/gr3.8tests/example_corr_est.py", line 295, in __init__
    self.digital_corr_est_cc_0 = digital.corr_est_cc(modulated_sync_word, sps, mark_delay, 0.999, digital.corr_est_cc.THRESHOLD_ABSOLUTE)
AttributeError: 'function' object has no attribute 'THRESHOLD_ABSOLUTE'

Replacing the last parameter in this function call by 0 or 1 ( the possible values of the enum ) the generated code runs.

dl1ksv, 2019-05-07:
The last issue would be fixed with #2471

mbr0wn, 2019-06-19:
@dl1ksv Merged a version of your PR, closing this.

closed 2019-06-19 by mbr0wn