gnuradio/gnuradio #2472: Converting XML flowgraphs in GNU Radio Companion when using GNU Radio 3.8

tferrett, 2019-05-07:
I'm migrating XML-based flowgraphs to GNU Radio 3.8 from older versions, however, conversion fails when opening the flowgraphs in Companion.  

My environment is Ubuntu 18.04 with GNU Radio 3.8 installed by following the instructions in the official readme at
  https://github.com/gnuradio/gnuradio/README.md
and installing Ubuntu packages as needed.

To reproduce the error, start Companion and open an XML-based flowgraph.  For example, the following flowgraph from the 3.7 maintenance branch
  https://github.com/gnuradio/gnuradio/blob/maint-3.7/gr-analog/examples/noise_power.grc
should produce the error.  The Companion console will display the following message when attempting to open
  >>> Converting from XML
  Error: expected str, bytes or os.PathLike object, not StreamReaderWriter
  >>> Failure

tferrett, 2019-05-07:
Self-diagnosed, needed to install Ubuntu package
  python3-lxml

skoslowski, 2019-05-08:
Your work-around is indeed correct. However, I would still like for the converter to work with the buildin xml support, so I am reopening this.

closed 2019-07-28 by marcusmueller

cloudlakecho, 2020-05-14:
I also have a problem in opening some GRC files. I could open some, but some I couldn't open. 

The Companion crashed without warning message (this is from Terminal):
```
.
.
.
" loaded from
  /home/roman/gnuradio/share/gnuradio/grc/blocks/zeromq_sub_source.block.yml
overwritten by
  /home/roman/gnuradio/share/gnuradio/grc/blocks/zeromq_sub_source.block.yml
<<< Welcome to GNU Radio Companion 3.8.1.0 >>>

Block paths:
	/home/roman/gnuradio/share/gnuradio/grc/blocks
	/home/roman/gnuradio/share/gnuradio/grc/blocks
Warning: restarting the docstring loader (crashed while loading 'qtgui_bercurve_sink')
Warning: restarting the docstring loader (crashed while loading 'qtgui_const_sink_x')
Warning: restarting the docstring loader (crashed while loading 'qtgui_edit_box_msg')
Warning: restarting the docstring loader (crashed while loading 'qtgui_freq_sink_x')
Warning: restarting the docstring loader (crashed while loading 'qtgui_histogram_sink_x')
Warning: docstring loader crashed too often

Loading: "/home/roman/Documents/USRP_SinkSource.grc"
>>> Converting from XML
/usr/lib/python3.6/importlib/_bootstrap.py:219: ImportWarning: can't resolve package from __spec__ or __package__, falling back on __name__ and __path__
  return f(*args, **kwds)
>>> Done

Loading: "/home/roman/Documents/CustomBlock.grc"
>>> Converting from XML
anom@jetson:
```

I checked `python3-lxml`, and it is already installed. 
The problematic file saved in Gnuradio verion 3.7.13.4

My environment
 Gnuradio version: 3.8.1.0
 gcc and g++ version: 7.5.0
 cmake version: 3.10.2
 OS: Ubuntu 18.04.4 LTS
 Machine: Nvidia Xavier (CPU: ARMv8 Processor rev 0 (v8l))

Please, help me.

Thanks.