gnuradio/gnuradio #2473: android support

copypasteearth, 2019-05-10:
hello I am trying to use gnuradio in android but the link to the tarbal goes to a 404. I am using this page as a reference https://wiki.gnuradio.org/index.php/Android and I do not know how to get the required android ndk-build parts of the application.  I have installed gnuradio and I have the companion and all that. is there a folder in that that I can point to the files in the jni part of the android application examples

thank you for your time

noc0lour, 2019-05-10:
Hi, happy to hear you are trying to get GNU Radio to work on Android. The missing tarball was created with this script https://github.com/trondeau/grand-build/. But the script is also outdated and to create a working build environment you need to invest some time to update the script or rebuild the dependencies on your own.

ValZapod, 2020-01-14:
Actually @bastibl is doint it. https://twitter.com/rtlsdrblog/status/1196629043694559233?s=19

NeuroForLunch, 2021-06-24:
Has this issue been resolved or become stale? If so please close it (mods).

closed 2021-06-26 by copypasteearth