gnuradio/gnuradio #2478: GRC Qt GUI & Python 3

dine909, 2019-05-14:
Gnuradio is a great project - It's ageing very quickly though.

There's a clear issue with:

* Deprecating Python 2
* Deprecating WX

But rarely mentioned is:

* Migrating GRC GUI into QT (C++ or Python3)

My thoughts are a that a concerted effort could nip the former if the latter was carried out in a major version release, i.e. GRC build in QT C++ or Python3.

I'd be willing to get involved in a major re-write.

noc0lour, 2019-05-14:
What's your issue with deprecation of Python2 & WX?  Python2 will be still available in GNU Radio 3.8 and WX is already removed.

Migrating GRC to Qt is a big project and it will definitely need some discussion if it would be worth it.

dine909, 2019-05-14:
> What's your issue with deprecation of Python2 & WX? Python2 will be still available in GNU Radio 3.8 and WX is already removed.

I meant the issue is it's still default - I have no issue with either being available, or not in 3.8. 

> Migrating GRC to Qt is a big project and it will definitely need some discussion if it would be worth it.

Certainly yes.  Where is the best place to start the discussion?

777arc, 2019-05-19:
I wouldn't mind getting involved as well, in GRC's port from pygtk to pyqt.  I assume we would keep it in Python, to minimize the amount of code that needs to be touched.

777arc, 2019-05-19:
Then again, does PyGI+GTK 3.0 actually lack support or is being depreciated?  Their git seems relatively active, I believe this is it: https://gitlab.gnome.org/GNOME/pygobject

Might be better off spending the effort adding features to the existing GRC

dine909, 2019-05-19:
>  I assume we would keep it in Python, to minimize the amount of code that needs to be touched.

Whilst it makes sense in theory to re-use the code, there isn't a lot of application specific code in there in the first instance.  It's mainly GUI bindings / generator.

However I think yes, keeping it in Python is a good idea for ease of development.  It's not like it's processor intensive.

> Might be better off spending the effort adding features to the existing GRC

My main motivation notes are (in no order, in no particular contexts):
* PyGtk looks dated, PyQT is orderly on every OS and supports more modern GUI mechanisms.
* Slick mime drag and drop implementation would be very useful
* Multi-select property editor
* Focus issues in non-native OS (Mac OS) are a real pain, triple clicking is tiring.
* Possibility for multi-level editing (drilling into sub-blocks)
* Canvas zooming, block resize, flowing auto-routed connectors.
* Currently gets a sluggish when working in large projects. (End up in a right click negative feedback loop often)
* Many more silly bits that would make a huge difference to the user experience.

noc0lour, 2019-05-20:
And removing GTK+/PyGTK in favor of  Qt isn't bad as well.

dkozel, 2019-10-01:
@dine909, with GNU Radio 3.8.0.0 released, WX entirely removed, and Python 2 disabled (full removal in progress) on the `master` branch now seems like a good time to pickup the discussion of moving GRC to QT. Are you still interested in working on this? If so I'd like to pull you into the discussion.

michaelld, 2019-10-01:
I'd love to see us move GRC to Qt! The resulting executable would look native on the target OS, and could be made more consistent across OSs (e.g., drag & drop, multi-clicking, etc) since those are handled by Qt and abstracted away from the OS. We've discussed this in the past multiple times with no resolution ... and thus the default has been using Python and GTK of some sort. +1 from me to finally make this so!

dine909, 2019-10-02:
@dkozel I am knee deep in a project at the moment but hope to be more free in a month or so - I am happy to join the discussion on how to move forward and piece a team together - 

Who's in?!

777arc, 2019-10-02:
Yeah I'll also have more free time in a month from now

dkozel, 2019-10-03:
Thanks @dine909 and @777arc. I'm going to reach out to @skoslowski to get the conversation kicked off. He's looked at it in the past and my understanding is that there were technical reasons the port was not done before.

michaelld, 2019-10-03:
I'll keep watching for a PR & will help with testing. I just don't have the work BW right now to do anything more useful!

skoslowski, 2019-12-05:
Yes, a Qt port of GRC would be awesome! I am all for it.

This has been attempted before, of course. My first naive attempt failed: I hadn't worked on the code base at all before. The GTK parts were all over and really hard to separate out. Nowadays the code is refactored. ``grc.core`` works stand-alone without any GTK (--> ``grcc``). A second attempt was made by never really got anywhere lacking resources (for python3, a lot of effort went into replacing Cheetah, so we opted to port from pygtk to pygobject instead of qt).

Building a new GUI (even alongside the current one) is very doable and I am happy to support this in way I can!

@sdh11 ready for a third attempt =)

dkozel, 2019-12-05:
@dine909 and @777arc there you have it. We're in a reasonably good place with the frontend of GRC already somewhat isolated from the core functionality. Are you still interested in working on this?  Sebastian is happy to support the development as am I, for what it's worth. I suspect that a few other people will be interested in being involved in testing and doing smaller tasks, but we need a developer to get this going.

777arc, 2019-12-07:
Let me go ahead and spend a few hours seeing how far I can get, I'll start by just trying to re-create the list of blocks pane in pyqt5.  That way I can get a feel for how much work it will be to do the whole thing.

777arc, 2019-12-19:
Update- it was pretty easy to get the list of blocks displayed in a tree, see below, but what I also wanted to do was have python print out the name/id of the block when I drag it from the list, and I just couldn't get that part working, I must be doing something wrong with the MIME stuff.  All the example code is specific to QFileSystemModel, seems like that's the only thing people are using tree + drag/drop for. 

[image]

haakov, 2020-01-04:
I would also be interested in getting involved with this :)

sdh11, 2020-01-06:
@skoslowski Sure, I'm up for it. I have been wanting to get back into GRC development. I may also still have the old QT development branch somewhere.

777arc, 2020-01-06:
Well, since @haakov and myself are currently actively creating a new version using pyqt5 from scratch, we should at least get a Slack channel going to communicate efforts, and not duplicate stuff.  I'll create the channel now and invite whoever I can find the Slack names of.

NeuroForLunch, 2021-06-24:
Seems like this issue can be closed

dkozel, 2021-06-26:
Development of the GRC Qt port is happening on the [gnuradio/feature-grc-qt](https://github.com/gnuradio/gnuradio/tree/feature-grc-qt) branch and being tracked on the [Qt GRCo Project](https://github.com/gnuradio/gnuradio/projects/7).

@dine909 Any chance you're interested in looping back onto this?

There's also a chat room for dev discussions.
https://chat.gnuradio.org/#/room/#grc-dev:gnuradio.org