gnuradio/gnuradio #2485: UNdefined reference to gr::digital::constellation::

9a39a3, 2019-05-15:
I am trying to add a new function member to the constellation class, but during the compile
I get this error: 
/usr/bin/ld  libgnuradio-digital-3.8git.so.0.0 undefined reference to gr::digital:constellation::myfunction
****____constellation.h:**
//
 class DIGITAL_API constellation
      : public boost::enable_shared_from_this<constellation>
    {
    public:
   ...
      **virtual void myfunction(const gr_complex x, unisgned int y);**
    ...
   }
  class DIGITAL_API constellation_custom : public constellation
    {
    public:
      typedef boost::shared_ptr<constellation_custom> sptr;
      static sptr make();
      ~constellation_custom();
      ...
      **void myfunction(const gr_complex x, unisgned int y);**
     ...
     protected:
      constellation_custom();
    }

****__constellation.cc:**
//
...
    constellation_custom::sptr
    constellation_custom::make()
    {
    return constellation_custom::sptr(new constellation_custom());
    }
  constellation_custom::constellation_custom()
  {
   }
  constellation_custom::~constellation_custom(){};

**void myfunction(const gr_complex x, unisgned int y);**
 {...
 }

michaelld, 2019-05-16:
A few thoughts:
(1) In the actual method implementation, you have to specify the class scope, otherwise the method is a function defined in the current namespace.

(2) You have a trailing ";" after the method implementation prototype that needs to be removed.

(3) If you're not defining the method in the base class, then you need to make it pure virtual by appending "= 0" after the prototype.

Some advice: Keep reading about c++ & how various parts are implemented and work between inherited classes. That's the trick in a lot of how GR does its c++ implementation.

Hope this is useful!

michaelld, 2019-05-16:
I'd advise you to create a GIST or whatever in your personal GITHUB or GITLAB or whatever space & then send a note to the GR list about what you're trying to do; you can refer to this issue if you want to, but note the same things you do here in your online posting & in brief in your email to the list.

michaelld, 2019-05-16:
I'm closing this issue since it has nothing to do directly with GR; nothing personal.

closed 2019-05-16 by michaelld