gnuradio/gnuradio #2491: Underflows with UHD USRP Sink

erikheinz, 2019-05-20:
I'm playing with gnuradio and an Ettus B210 device.
I'm new to gnuradio and have to use git master branch since my system is Python-3-only.

The issue: even the simplest flow graph generated by grc and containing UHD USRP Sink gives lots of 'U's if the sampling rate is larger than about 100k. Same with GUI disabled.
uhd_siggen and uhd_siggen_gui on the other hand work fine up to sampling rates of 30M and more..

My system: home grown Linux, gcc 7.4.0, Python 3.6.4, Qt 5.12.3, Boost 1.70.0, UHD 3.14.0, gnuradio master as of May 13th.

Any ideas? Thanks in advance.

Erik

michaelld, 2019-05-20:
Is _any_ data getting through when using GRC? If you track down the GR example "uhd_fft.py", does it work at all?

Related but for just Boost 1.70.0: How did you fixup GR to work with this new Boost? Did you install the Boost CMake find scripts && if so, how did you work around their issues within Volk, UHD, and GR?

erikheinz, 2019-05-20:
Sending/receiving works fine up to a sampling rate of about 100k, even simultaneously. At higher sampling  rates the 'U's are starting to appear. This happens as well if I start the Python code generated by GRC when GRC is not running.

uhd_fft seems to have some other issues for now:
  File "/usr/bin/uhd_fft", line 510, in <module>
    main()
  File "/usr/bin/uhd_fft", line 488, in main
    tb = uhd_fft(args)
  File "/usr/bin/uhd_fft", line 226, in __init__
    self._qtgui_waterfall_sink_x_0_win = sip.wrapinstance(self.qtgui_waterfall_sink_x_0.pyqwidget(), Qt.QWidget)
TypeError: wrapinstance() argument 2 must be sip.wrappertype, not sip.wrappertype

As for the building: I used a vanilla Boost installation and fixed things manually after running cmake. This is an ugly hack, of course, but I saw no other chance. Fixing a cmake build system is beyond my knowledge. Most of the work was adding the boost libraries to the linking commands. I linked the uhd library against the boost libraries which might have made things easier for building gnuradio. For the "no member named 'get_io_service' problem I used the fix from #2451.

michaelld, 2019-05-20:
Interesting. A few things to note:

(1) GRC just executes the Python code it generates. Doing so manually is the same as what GRC does, for all practical purposes. Once GRC can generate and compile C++ code successfully, that might make a difference ... but until then, executing in GRC is the same as running the Python script.

(2) To fix the Boost CMake crap and make GR usable with 1.70.0:

(a) Use this fix instead < https://svnweb.freebsd.org/ports/head/comms/gnuradio/files/patch-gr-blocks_lib_socket__pdu__impl.cc?view=markup&pathrev=499093 >. It's more portable and more stable than the one you note.

(b) Just get rid of the Boost CMake stuff, post install -- e.g., `cd /usr/local/lib && rm -rf $(/bin/ls | grep -i boost | grep 1.70.0)` ! It wasn't ready for release, yet Boost released it & made its install the default && now it's messing up other projects' use of Boost! [They have a patch in boost_install already that fixes it to work better < https://github.com/boostorg/boost_install/blob/develop/BoostConfig.cmake >; still not quite there yet but certainly good enough for what projects using this Boost will require.]

(3) That's an odd error executing "uhd_fft" ... maybe it's a GR 3.8 issues? Can you go into your GR38 build directory & "sudo make uninstall". Then grab the latest GR 3.7.14.5 release & build & install it instead & see if that helps? If it doesn't then it's most likely a UHD issue since that's the common project between both GR's.

erikheinz, 2019-05-21:
Thanks for the useful information.

As for the UHD sampling rate bottleneck - I have found the reason. "Max Number of Output" was set to 1 in the GRC Options block.
With a value of 0, the underflows are gone.

Best regards,
Erik

closed 2019-05-21 by erikheinz

michaelld, 2019-05-21:
Interesting! Glad you found the issue & are up and going with UHD and GRC now!

zhanght34, 2022-04-16:
did you solve this problem？I also have an underflow problem.