gnuradio/gnuradio #2494: grc fails to build constellation object if constellation != variable constellation

dl1ksv, 2019-05-21:


dl1ksv, 2019-06-22:
Just open a new flowgraph in grc and add a Constellation object from Modulators . Then you get immediately:

ERROR:gnuradio.grc.core.FlowGraph:Failed to evaluate variable block variable_constellation_0
Traceback (most recent call last):
  File "/usr/local/gnuradio/lib/python3.7/dist-packages/gnuradio/grc/core/FlowGraph.py", line 226, in renew_namespace
    value = eval(variable_block.value, namespace, variable_block.namespace)
  File "<string>", line 1, in <module>
AttributeError: module 'gnuradio.digital' has no attribute 'constellation_type'

closed 2019-07-25 by marcusmueller

dj0abr, 2020-09-27:
the problem still exists if the output language is switched to C++. While it work in Python.

PR #2662 (2019-07-25): KeyError patch for variable blocks 
Together with #2605 this fixes #2494 and KeyErrors in other variable blocks
PR #2605 (2019-08-02): digital: constellation object yml error
value parameter field has invalid reference to constellation_type

Cleaned up the yml, but probably needs some insight into the
yml generator

fixes #2494
commit 86f3d3c1 (2019-08-02): digital: constellation object yml error

value parameter field has invalid reference to
constellation_type

Cleaned up the yml, but probably needs some insight into the
yml generator

fixes #2494