gnuradio/gnuradio #2498: gnuradio-companion is no longer installed?

stutys, 2019-05-23:
gnuradio-companion is not installed after successfully installing gnuradio from source, and nothing could be found by command 'locate gnurdaio-companion' .

gunradio-companion is not defaultly installed ? or could not install it by building from source?

michaelld, 2019-05-23:
You'll need to provide a little more information here for us to help you. GRC continues to be available when building GR from source. If you have GR 3.7 installed with GRC and haven't updated dependencies for GR 3.8, and try to build GR 3.8 from source, then GRC will be not be enabled because its dependencies won't be met.

stutys, 2019-05-23:
@michaelld Thanks a lot. I didn't update the dependencies, and built GR from source directly. I will try it again. I will give more details if some error encountered, thanks.

stutys, 2019-05-24:
@michaelld I updated the drivers, including uhd, soapysdr,soapyuhd and limesuite. And recompile gnuradio from source again. However, the command ```gnuradio-companion``` is still not avaliable.

I installed gnuradio from source followed by commands below,

```
$ git clone --recursive https://github.com/gnuradio/gnuradio.git
$ cd gnuradio && mkdir build && cd build
$ cmake ../
$ make -j5
$ make test
$ sudo make install
$ sudo ldconfig
```
and the last output lines of ```sudo make install``` is, 
```
...
-- Installing: /usr/local/lib/cmake/gnuradio/gnuradio-waveletConfig.cmake
-- Installing: /usr/local/lib/python3.6/dist-packages/gnuradio/wavelet/_wavelet_swig.so
-- Set runtime path of "/usr/local/lib/python3.6/dist-packages/gnuradio/wavelet/_wavelet_swig.so" to ""
-- Installing: /usr/local/lib/python3.6/dist-packages/gnuradio/wavelet/wavelet_swig.py
-- Installing: /usr/local/lib/python3.6/dist-packages/gnuradio/wavelet/wavelet_swig.pyc
-- Installing: /usr/local/lib/python3.6/dist-packages/gnuradio/wavelet/wavelet_swig.pyo
-- Installing: /usr/local/include/gnuradio/swig/wavelet_swig.i
-- Installing: /usr/local/include/gnuradio/swig/wavelet_swig_doc.i
-- Installing: /usr/local/lib/python3.6/dist-packages/gnuradio/wavelet/__init__.py
-- Installing: /usr/local/lib/python3.6/dist-packages/gnuradio/wavelet/__init__.pyc
-- Installing: /usr/local/lib/python3.6/dist-packages/gnuradio/wavelet/__init__.pyo
-- Installing: /usr/local/lib/pkgconfig/gnuradio-zeromq.pc
-- Installing: /usr/local/include/gnuradio/zeromq/api.h
-- Installing: /usr/local/include/gnuradio/zeromq/pub_sink.h
-- Installing: /usr/local/include/gnuradio/zeromq/pub_msg_sink.h
-- Installing: /usr/local/include/gnuradio/zeromq/sub_source.h
-- Installing: /usr/local/include/gnuradio/zeromq/sub_msg_source.h
-- Installing: /usr/local/include/gnuradio/zeromq/pull_source.h
-- Installing: /usr/local/include/gnuradio/zeromq/pull_msg_source.h
-- Installing: /usr/local/include/gnuradio/zeromq/push_sink.h
-- Installing: /usr/local/include/gnuradio/zeromq/push_msg_sink.h
-- Installing: /usr/local/include/gnuradio/zeromq/rep_sink.h
-- Installing: /usr/local/include/gnuradio/zeromq/rep_msg_sink.h
-- Installing: /usr/local/include/gnuradio/zeromq/req_source.h
-- Installing: /usr/local/include/gnuradio/zeromq/req_msg_source.h
-- Installing: /usr/local/lib/libgnuradio-zeromq.so.3.8tech-preview-373-g8a7214c7
-- Installing: /usr/local/lib/libgnuradio-zeromq.so.3.8git
-- Set runtime path of "/usr/local/lib/libgnuradio-zeromq.so.3.8tech-preview-373-g8a7214c7" to ""
-- Installing: /usr/local/lib/libgnuradio-zeromq.so
-- Installing: /usr/local/lib/cmake/gnuradio/gnuradio-zeromqTargets.cmake
-- Installing: /usr/local/lib/cmake/gnuradio/gnuradio-zeromqTargets-release.cmake
-- Installing: /usr/local/lib/cmake/gnuradio/gnuradio-zeromqConfig.cmake
-- Installing: /usr/local/lib/python3.6/dist-packages/gnuradio/zeromq/_zeromq_swig.so
-- Set runtime path of "/usr/local/lib/python3.6/dist-packages/gnuradio/zeromq/_zeromq_swig.so" to ""
-- Installing: /usr/local/lib/python3.6/dist-packages/gnuradio/zeromq/zeromq_swig.py
-- Installing: /usr/local/lib/python3.6/dist-packages/gnuradio/zeromq/zeromq_swig.pyc
-- Installing: /usr/local/lib/python3.6/dist-packages/gnuradio/zeromq/zeromq_swig.pyo
-- Installing: /usr/local/include/gnuradio/swig/zeromq_swig.i
-- Installing: /usr/local/include/gnuradio/swig/zeromq_swig_doc.i
-- Installing: /usr/local/lib/python3.6/dist-packages/gnuradio/zeromq/__init__.py
-- Installing: /usr/local/lib/python3.6/dist-packages/gnuradio/zeromq/rpc_manager.py
-- Installing: /usr/local/lib/python3.6/dist-packages/gnuradio/zeromq/probe_manager.py
-- Installing: /usr/local/lib/python3.6/dist-packages/gnuradio/zeromq/__init__.pyc
-- Installing: /usr/local/lib/python3.6/dist-packages/gnuradio/zeromq/rpc_manager.pyc
-- Installing: /usr/local/lib/python3.6/dist-packages/gnuradio/zeromq/probe_manager.pyc
-- Installing: /usr/local/lib/python3.6/dist-packages/gnuradio/zeromq/__init__.pyo
-- Installing: /usr/local/lib/python3.6/dist-packages/gnuradio/zeromq/rpc_manager.pyo
-- Installing: /usr/local/lib/python3.6/dist-packages/gnuradio/zeromq/probe_manager.pyo
-- Installing: /usr/local/share/gnuradio/examples/zeromq/zeromq_pubsub.grc
-- Installing: /usr/local/share/gnuradio/examples/zeromq/zeromq_pushpull.grc
-- Installing: /usr/local/share/gnuradio/examples/zeromq/zeromq_reqrep.grc
-- Installing: /usr/local/share/gnuradio/examples/zeromq/zmq_msg.grc
-- Installing: /usr/local/share/gnuradio/examples/zeromq/zmq_stream.grc
```
[... 4 lines cut ...]
  -h [ --help ]         print help message
  --prefix              print GNU Radio installation prefix
  --sysconfdir          print GNU Radio system configuration directory
  --prefsdir            print GNU Radio preferences directory
  --userprefsdir        print GNU Radio user preferences directory
  --prefs               print GNU Radio preferences
  --builddate           print GNU Radio build date (RFC2822 format)
  --enabled-components  print GNU Radio build time enabled components
  --cc                  print GNU Radio C compiler version
  --cxx                 print GNU Radio C++ compiler version
  --cflags              print GNU Radio CFLAGS
  -v [ --version ]      print GNU Radio version
```
All seams right, but i can not launch GRC by command ```gnuradio-companion```,
```
$ gnuradio-companion
The program 'gnuradio-companion' is currently not installed. You can install it by typing:
sudo apt install gnuradio
```
[... cut ...] enable-components,
```
$ gnuradio-config-info --enabled-components
python-support;testing-support;volk;doxygen;sphinx;gnuradio-runtime;gr-ctrlport;gr-blocks;gr-fec;gr-fft;gr-filter;gr-analog;gr-digital;gr-dtv;gr-audio;* alsa;* oss;gr-comedi;gr-channels;gr-trellis;gr-uhd;gr-utils;gr-video-sdl;gr-vocoder;gr-wavelet;gr-zeromq
```
whether my compiling option settings was wrong,  gnuradio-companion should be included manully? or some other methods could be provided to launch ```gnuradio-companion```?

dkozel, 2019-05-24:
Since gr-qtgui is also missing my first guess is that the appropriate QT libraries are missing. What OS are you building on? Guessing it's Ubuntu here is a page listing the dependencies.
https://wiki.gnuradio.org/index.php/UbuntuInstall#Install_Dependencies

If using those and rerunning cmake does not show gnuradio-companion as an enabled component (you don't need to actually build, just look at the end of the cmake output), then please paste the full output of running cmake, including the actual cmake command, here.

michaelld, 2019-05-24:
I'm with @dkozel here: it looks like some of the Qt installs are missing; otherwise, the list of installed components would contain `gr-qtgui`, and with that you'd also probably get GRC since in GR3.8 it now uses Qt5-related projects. Do look through the Ubuntu install dependencies link & see if that helps clear up which installs you need.

noc0lour, 2019-05-24:
Can you check and/or paste the output of your `cmake` run? Probably you have missing dependencies for gnuradio-companion.

stutys, 2019-05-24:
Thanks @michaelld @dkozel @noc0lour 
The output of ```cmake``` is,
```
$ cmake -DCMAKE_INSTALL_PREFIX=/opt/gnuradio ../
-- Build type not specified: defaulting to release.
-- Build type set to Release.
-- Extracting version information from git describe...
-- Compiler Version: cc (Ubuntu 5.4.0-6ubuntu1~16.04.11) 5.4.0 20160609
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
-- Compiler Flags: /usr/bin/cc:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
/usr/bin/c++:::-O3 -DNDEBUG  -fvisibility=hidden -Wsign-compare -Wall -Wno-uninitialized
-- ADDING PERF COUNTERS
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   date_time
--   program_options
--   filesystem
--   system
--   regex
--   thread
--   unit_test_framework
--   chrono
--   atomic
-- User set python executable /home/epson/anaconda3/bin/python3
-- Found PythonLibs: /home/epson/anaconda3/lib/libpython3.6m.so (found suitable exact version "3.6.8") 
-- Found PythonLibs: /home/epson/anaconda3/lib/libpython3.6m.so (found suitable version "3.6.8", minimum required is "2.7") 
-- Python checking for six - python 2 and 3 compatibility library - found
-- 
-- Checking for module SWIG
-- Found SWIG version 3.0.8.
-- 
-- The build system will automatically enable all components.
-- Use -DENABLE_DEFAULT=OFF to disable components by default.
-- 
-- Configuring python-support support...
--   Dependency PYTHONLIBS_FOUND = TRUE
--   Dependency SWIG_FOUND = TRUE
--   Dependency SWIG_VERSION_CHECK = TRUE
--   Dependency SIX_FOUND = TRUE
--   Enabling python-support support.
--   Override with -DENABLE_PYTHON=ON/OFF
-- 
-- Configuring testing-support support...
--   Dependency Boost_FOUND = 1
--   Enabling testing-support support.
--   Override with -DENABLE_TESTING=ON/OFF
-- 
-- Configuring VOLK support...
-- Build type set to Release.
-- Extracting version information from git describe...
-- 
-- Python checking for python >= 2.7
-- Python checking for python >= 2.7 - found
-- 
-- Python checking for mako >= 0.4.2
-- Python checking for mako >= 0.4.2 - found
-- 
-- Python checking for six - python 2 and 3 compatibility library
-- Python checking for six - python 2 and 3 compatibility library - found
-- Boost version: 1.58.0
-- Found the following Boost libraries:
--   filesystem
--   system
-- Checking for module 'orc-0.4 > 0.4.11'
--   No package 'orc-0.4' found
-- orc files (missing: ORC_LIBRARY ORC_INCLUDE_DIR ORCC_EXECUTABLE) 
-- QA Testing is enabled.
--   Modify using: -DENABLE_TESTING=ON/OFF
[... 775 lines cut ...]

    runtime library [libgmpxx.so.4] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/epson/anaconda3/lib
    runtime library [libgmp.so.10] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/epson/anaconda3/lib
    runtime library [libzmq.so.5] in /usr/lib/x86_64-linux-gnu may be hidden by files in:
      /home/epson/anaconda3/lib

  Some of these libraries may not be found correctly.
Call Stack (most recent call first):
  cmake/Modules/GrSwig.cmake:133 (swig_add_library)
  gr-zeromq/swig/CMakeLists.txt:34 (GR_SWIG_MAKE)

-- Generating done
-- Build files have been written to: /home/epson/Localization/gnuradio/build
```
you are right , it reveals 
```
-- ######################################################
-- # Gnuradio disabled components                        
-- ######################################################
--   * gnuradio-companion
--   * gr-qtgui
```
i'm trying to install the dependencies, like ```libqwt-qt5-dev libqt5opengl5-dev python3-pyqt5 python3-yaml```， but it cannot be installed by ```apt-get install```. Anyway, that's not the gnuradio's issue.

stutys, 2019-05-24:
Thanks, guys. I finally installed successfully by pybombs.

closed 2019-05-24 by stutys

michaelld, 2019-05-24:
:+1: glad you got it working && thanks for actively reporting here!

tinyxuyan, 2020-07-03:
I got a similar error as the author.
The output of the command: `gnuradio-config-info --enabled-components`
is `testing-support;python-support;volk;doxygen;sphinx;gnuradio-runtime;gr-ctrlport;gr-blocks;gr-fec;gr-fft;gr-filter;gr-analog;gr-digital;gr-dtv;gr-audio;* alsa;* oss;gr-channels;gr-qtgui;gr-trellis;gr-utils;gr-video-sdl;gr-vocoder;gr-wavelet;gr-zeromq`

It seems like that the gr-qtgui is successfully installed. However, when I type 'gnuradio-companion' in the terminal, it outputs that the command is not found.

When I try to install gnu-radio through pybombs, it fails too.
The commands for installation using pybombs is 
`sudo -H pip3 install PyBOMBS `  
`pybombs auto-config   `
`pybombs recipes add-defaults  `  
`pybombs prefix init ~/gnuradio -R gnuradio-default`
During the waiting process, the terminal outputs error like https://github.com/gnuradio/gnuradio/issues/2498#issuecomment-495610830.

good-point-lab, 2022-01-19:
Building and installing from the source on RaspberryPi OS following the instructions and observing the same problem - there is no 'gnuradio-companion' available.   Binary install works as expected, but gives 3.8.2.  I need at least 3.9 to build the current 'gr-osmosdr' from the source.   Please advice!

velichkov, 2022-01-20:
Hi @good-poin-lab,

Better post a message in the gnuradio's [mailing list](https://wiki.gnuradio.org/index.php/MailingLists) or [chat.gnuradio.org](https://chat.gnuradio.org/) and avoid posting in long-closed github issues. 

Provide the full output of "cmake -DENABLE_GRC=ON .." as most probably some dependency is missing.

Read also:

https://mail.gnu.org/archive/html/discuss-gnuradio/2022-01/msg00068.html
https://mail.gnu.org/archive/html/discuss-gnuradio/2022-01/msg00007.html
https://mail.gnu.org/archive/html/discuss-gnuradio/2021-11/msg00152.html

noQuitter, 2024-03-05:
I had the same problem by installing gnuradio from source. It appears after the steps "cmake"-->''make"-->"make install" that every thing installed successfully but gnuradio-companion could not be found. I have moved on and installed it through apt install gnuradio. But i think this problem is worth looking into.