gnuradio/gnuradio #2504: Add docs for semantic versioning, rename CMake vars

mbr0wn, 2019-05-23:
See discussion here: https://github.com/gnuradio/gnuradio/pull/2487

UHD SemVer Docs: https://github.com/EttusResearch/uhd/blob/master/host/docs/uhd_semvar.dox

777arc, 2019-06-21:
Looks like Marcus created the docs - https://github.com/gnuradio/gnuradio/blob/master/VERSIONING  
Are they good to go?  

As far as the CMake variables, did we want to use Ettus' format and just call them:
VERSION_MAJOR
VERSION_API
VERSION_ABI
VERSION_PATCH
?

Any other docs or CMake vars I'm missing?

marcusmueller, 2019-07-15:
Will have to check whether the distro packaging scripts read these. Noc0lour confirms our CI doesn't, so that's fine.

noc0lour, 2019-07-15:
:+1: And if anything breaks we can fix it later ¯\_(ツ)_/¯

marcusmueller, 2019-07-15:
@noc0lour doesn't work like that with packaging; this is a bit more sensitive than code.

noc0lour, 2019-07-15:
It does, it's not like they publish packages without having a look at it.

noc0lour, 2019-07-15:
If you change something, do it, put it in changelogs. Of course there as a chance that this is totally stupid, but that's why we review stuff here.

marcusmueller, 2019-07-15:
@noc0lour call me paranoid, but as a software maintainer, it's my core responsibility to make sure downstream doesn't break. This is a change where I need to do 15 minutes of due diligence.

Simple philosophy: Don't be aggressive in code changes when the benefits are more beauty in four variable names and the downsides would be breaking e.g. the fedora build chain. 

So, that's my review here: won't merge until I've checked the major distro's specs/.desc/.ebuilds/aur.  Again, don't forget that I'm not doing this for my own fun alone, but because I want GNU Radio to be usable for as many people as possible. Not breaking distros ranks comparatively high in my priority list, due to that.

marcusmueller, 2019-07-15:
Checked that none of the major distros will have an issue with this.

PR #2624 (2019-07-17): Renaming the VERSION_INFO_* to VERSION_ macros consistent w/ VERSIONING
closes #2504
closed 2019-07-17 by marcusmueller