gnuradio/gnuradio #2506: qa_qtgui test failing

pianoman730, 2019-05-23:
After much deliberation of trying to get the qt gui stuff to be enabled, I was able to build and run the tests, but the `qa_qtgui` test failed. Running `ctest -V -R qa_qtgui` gives: 
```
UpdateCTestConfiguration  from :/home/<user>/dev/gnuradio/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/home/<user>/dev/gnuradio/build/DartConfiguration.tcl
Test project /home/<user>/dev/gnuradio/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 345
    Start 345: qa_qtgui

345: Test command: /usr/bin/sh "/home/<user>/dev/gnuradio/build/gr-qtgui/python/qtgui/qa_qtgui_test.sh"
345: Test timeout computed to be: 10000000
345: Traceback (most recent call last):
345:   File "/home/<user>/dev/gnuradio/gr-qtgui/python/qtgui/qa_qtgui.py", line 24, in <module>
345:     from gnuradio import gr, gr_unittest, qtgui
345:   File "/home/<user>/dev/gnuradio/gr-qtgui/python/qtgui/__init__.py", line 38, in <module>
345:     from .range import Range, RangeWidget
345:   File "/home/<user>/dev/gnuradio/build/gr-qtgui/python/qtgui/range.py", line 24, in <module>
345:     from PyQt5 import Qt, QtCore, QtWidgets
345: ImportError: /lib64/libQt5Core.so.5: version `Qt_5.12' not found (required by /home/<user>/.local/lib/python3.6/site-packages/PyQt5/QtCore.so)
1/1 Test #345: qa_qtgui .........................***Failed    0.23 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.26 sec

The following tests FAILED:
	345 - qa_qtgui (Failed)
Errors while running CTest
```

I'm not sure why this is failing, as I have these installed at the right version:
```
python36-qt5.x86_64                     5.12.1-2.el7                   @epel    
python36-qt5-base.x86_64                5.12.1-2.el7                   @epel    
python36-qt5-devel.x86_64               5.12.1-2.el7                @epel
```
However, centos 7 does not have `qt5-qtbase.x86_64` as anything higher than version 5.9. Is there a reason this dependency would exist here making this test fail?

michaelld, 2019-06-11:
Is this issue still relevant? Seems very much like another we've recently worked through.

pianoman730, 2019-06-14:
Nope we're good. I was running it through ssh. I got onto the actual desktop environment and it passed.

closed 2019-06-14 by pianoman730

bencampionmax, 2019-12-10:
Hello,

I am having this problem. What fixed this issue?

velichkov, 2019-12-10:
Hi @bencampionmax,

> I am having this problem. What fixed this issue?

Are you running the tests on a headless system (no X11, Wayland, ... )? If so try running the tests under [xvfb](https://en.wikipedia.org/wiki/Xvfb)

```
xvfb-run ctest -V -R qa_qtgui
```
and if it still fails provide the full output of the above command.

cloudlakecho, 2020-05-28:
@velichkov Thanks for the suggestion. I have the same error and this is my output
```
roman@jetson-0423018054844:~/workarea-gnuradio/gnuradio/build$ xvfb-run ctest -V -R qa_qtgui
UpdateCTestConfiguration  from :/home/roman/workarea-gnuradio/gnuradio/build/DartConfiguration.tcl
UpdateCTestConfiguration  from :/home/roman/workarea-gnuradio/gnuradio/build/DartConfiguration.tcl
Test project /home/roman/workarea-gnuradio/gnuradio/build
Constructing a list of tests
Done constructing a list of tests
Updating test list for fixtures
Added 0 tests to meet fixture requirements
Checking test dependency graph...
Checking test dependency graph end
test 348
    Start 348: qa_qtgui

348: Test command: /bin/sh "/home/roman/workarea-gnuradio/gnuradio/build/gr-qtgui/python/qtgui/qa_qtgui_test.sh"
348: Test timeout computed to be: 9.99988e+06
348: Segmentation fault (core dumped)
1/1 Test #348: qa_qtgui .........................***Failed    0.91 sec

0% tests passed, 1 tests failed out of 1

Total Test time (real) =   0.95 sec

The following tests FAILED:
	348 - qa_qtgui (Failed)
Errors while running CTest

```

I did `cmake -DQWT_LIBRARIES=/usr/lib/libqwt.so ../` to make the qtgui enable and this command completed without an error.

Is there anything I miss? Thanks.

My environment
Gnuradio version: 3.8.0.0
gcc and g++ version: 7.5.0
cmake version: 3.10.2
OS: Ubuntu 18.04.4 LTS
Machine: Nvidia Xavier (CPU: ARMv8 Processor rev 0 (v8l))

velichkov, 2020-05-30:
Hi @cloudlakecho,

> I did cmake -DQWT_LIBRARIES=/usr/lib/libqwt.so ../ to make the qtgui enable and this command completed without an error.

> lrwxrwxrwx 1 root   20 Jan 12  2018 libqwt.so -> libqwt.so.6.1.3.abi1
> -rw-r--r-- 1 root 1.2M Jan 12  2018 libqwt.so.6.1.3.abi1
> lrwxrwxrwx 1 root   20 Jan 12  2018 libqwt.so.6abi1 -> libqwt.so.6.1.3.abi1

In Ubuntu 18.04 this qwt library is build with Qt4, to confirm this run `ldd libqwt.so | grep Qt`. For gnuradio 3.8 you need the Qt5 versions so install `libqwt-qt5-dev` package then go in the gnuradio's build directory, delete `CMakeCache.txt`, rerun the cmake without `-DQWT_LIBRARIES=` parameter and then rebuild and retest.

Also if you don't need `libqwt-dev` and `libqt4-dev` for something else you may consider uninstalling them to prevent conflicts.

cloudlakecho, 2020-06-01:
@velichkov Thanks for suggestion of recommended QT version.
 
I didn't know that Gnuradio 3.8 needed QT version 5...Where can I find the tool dependency requirement for Gnuradio 3.8 in the Internet? 

Regarding other QT version in my Xavier, I didn't mention in here, but I am also trying to install UHD program in Xavier. I was mimicking [this instruction for UHD and Gnuradio installation from source](http://blogspot.tenettech.net/2019/09/13/interfacing-software-define-radio-sdr-with-nvidia-nano-jetson-development-kit/) This instruction recommended other QT version.

velichkov, 2020-06-02:
Hi @cloudlakecho,

> I didn't know that Gnuradio 3.8 needed QT version 5...Where can I find the tool dependency requirement for Gnuradio 3.8 in the Internet?

You can find a list of all dependency packages in the [wiki](https://wiki.gnuradio.org/index.php/UbuntuInstall#Install_Dependencies).

> Regarding other QT version in my Xavier, I didn't mention in here, but I am also trying to install UHD program in Xavier. I was mimicking [this instruction for UHD and Gnuradio installation from [...]

This blog post is for an older gnuradio version and that's why they recommend installing an older QT version

    cd gnuradio
    git checkout v3.7.13.4

cloudlakecho, 2020-06-02:
@velichkov, thanks for the link and indication on blog post. 
I was actually aware the blog post for the Gnuradio version 3.7.

Meanwhile, I start from scratch (reinstall OS at Xavier) and installed Gnuradio 3.8 (in clean environment) using Pybombs followed the instruction from the git. 

Unfortunately, I am still not able to run the Gnuradio-Companion, Here is error message:
[image]

I think I set my environmental followed the web instruction.

Here is my `~/.bashrc`:
```
export PATH=$PATH:/usr/local/cuda-10.0/bin
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/cuda-10.0/lib64:/home/nvidia/gnuradio/lib:
export PYTHONPATH=$PYTHONPATH:/home/nvidia/gnuradio/lib/python3/dist-packages/gnuradio:/homr/nvidia/gnuradio/lib/python3.6/site-packages/uhd:
```

Regarding dependent tools, especially QT, it looks like correct now:
```
nvidia@x02:~$ ls /usr/lib | grep qt
libqwt-qt5.so
libqwt-qt5.so.6
libqwt-qt5.so.6.1
libqwt-qt5.so.6.1.3
qt5
```

My environment
        OS: Ubuntu 18.04.4 LTS
        Machine: Nvidia Xavier (CPU: ARMv8 Processor rev 0 (v8l))

Please, give me some suggestions, thanks.

cloudlakecho, 2020-06-03:
@velichkov, I found the cause of my error. I put incorrect and miss one path in my `~/.bashrc` (and `setup_env.sh` I guess):

Here are correct my `~/.bashrc`:
```
export PATH=$PATH:/usr/local/cuda-10.0/bin:/home/nvidia/gnuradio/bin
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/cuda-10.0/lib64:/usr/local/lib:/home/nvidia/gnuradio/lib:
export PYTHONPATH=$PYTHONPATH:/usr/local/lib/python3.6/dist-packages:/home/nvidia/gnuradio/lib/python3/dist-packages:
```

And `setup_env.sh`:
```
export PATH="$PATH:/home/nvidia/gnuradio/bin:"
export PYTHONPATH="$PYTHONPATH:/home/nvidia/gnuradio/lib/python3/dist-packages:"
export LD_LIBRARY_PATH="$LD_LIBRARY_PATH:/home/nvidia/gnuradio/lib:"
export LIBRARY_PATH="$LIBRARY_PATH:/home/nvidia/gnuradio/lib:"
export PKG_CONFIG_PATH="$PKG_CONFIG_PATH:/home/nvidia/gnuradio/lib/pkgconfig:/home/nvidia/gnuradio/lib64/pkgconfig:"
````

Thanks for your suggestion!

sdjkjsdh, 2022-08-07:
> Nope we're good. I was running it through ssh. I got onto the actual desktop environment and it passed.

This problem dismissed when use the actual desktop environment too.
It seems that the remote ssh actually has this problem. I have met similar qt issue when I use ssh to start gedit